RobotLocomotion/drake #7519: Import pydrake works with python3, but not python2

hongkai-dai, 2017-11-27:
I tried to set the `PYTHONPATH` variable to the folder in drake, as suggested in http://drake.mit.edu/python_bindings.html. I can then do `import pydrake` using python3, but not python2.

Here is the screenshot
[image]

hongkai-dai, 2017-11-27:
cc @weiqiao

EricCousineau-TRI, 2017-11-27:
Sorry about that @weiqiao! The instructions we have are not exactly explicit on how to use the Python bindings.

I'd expect that `pydrake` would not work just adding `bazel-bin/...` to `PYTHONPATH`. (TBH, it's surprising that the Python3 imports work. Will check into that.)
At present, the main (but undocumented) way to do this is to first build and install Drake, and then use this setup to leverage Python bindings.

I have not yet tested this on my machine, but this should be roughly what you need at this point:
```bash
cd drake
# Install to a local directory. "~+" means current working directory.
mkdir -p build/install  # Or whatever you want to call it
bazel run //:install -- ~+/build/install
# Update environment variables; solibs and Python.
export LD_LIBRARY_PATH=~+/build/install/lib:${LD_LIBRARY_PATH}
export PYTHONPATH=~+/build/install/lib/python2.7/site-packages:${PYTHONPATH}
# Now try to import pydrake. (Presently supported in Python2. Python3 may work.)
python2 -c 'import pydrake; print(pydrake.__file__)'
```

If you need a slightly quicker but much hackier way, let me know, but please try this first.

The main functionality of `MathematicalProgram`, `RigidBodyTree`, and symbolics is already exposed. The System framework is not yet exposed, but will be soon.

hongkai-dai, 2017-11-27:
Thanks @EricCousineau-TRI , it would be great if we could add this setup instruction to http://drake.mit.edu/python_bindings.html

fbudin69500, 2017-11-28:
What @EricCousineau-TRI wrote is correct but can be simplified a little bit. It should not be necessary to add:
```export LD_LIBRARY_PATH=~+/build/install/lib:${LD_LIBRARY_PATH}
```
If it doesn't work without this line, there is a bug that I should correct...

jamiesnape, 2017-11-30:
See also https://github.com/RobotLocomotion/drake/issues/7507#issuecomment-347643528.

jamiesnape, 2017-11-30:
I would encourage people to use `cmake` instead of `bazel run //:install` where possible. We have options for GUROBI, MOSEK, and SNOPT support also.

EricCousineau-TRI, 2017-12-06:
Gotcha - will cater those instructions towards pointing to `CMAKE_INSTALL_PREFIX`.
That being said, I would still like to expose Bazel instructions, at least for Drake developers, if they are not consuming it via CMake.

jamiesnape, 2017-12-06:
There are a lot of caveats to using that approach, so maybe you can document it internally only.

EricCousineau-TRI, 2018-01-29:
Closing, per discussion in #7891.

closed 2018-01-29 by EricCousineau-TRI