RobotLocomotion/drake #7520: LeakSanitizer detects memory leaks in rgbd_camera_test and rgbd_renderer_test

jamiesnape, 2017-11-27:
```
=================================================================
==33204==ERROR: LeakSanitizer: detected memory leaks

Indirect leak of 112 byte(s) in 2 object(s) allocated from:
    #0 0x5619f4993b4e  (/home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-memcheck-lsan/_bazel_ubuntu/14b081bf825c73cab1654e615f0285b1/execroot/drake/bazel-out/clang-4.0-linux-dbg/bin/drake/systems/sensors/rgbd_camera_test+0x1673b4e)
    #1 0x7f5c8c1f9c1b  (<unknown module>)

Indirect leak of 112 byte(s) in 2 object(s) allocated from:
    #0 0x5619f4993b4e  (/home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-memcheck-lsan/_bazel_ubuntu/14b081bf825c73cab1654e615f0285b1/execroot/drake/bazel-out/clang-4.0-linux-dbg/bin/drake/systems/sensors/rgbd_camera_test+0x1673b4e)
    #1 0x7f5c8c1f9c32  (<unknown module>)

SUMMARY: LeakSanitizer: 224 byte(s) leaked in 4 allocation(s).
```

jamiesnape, 2017-12-01:
Turns out this is graphics driver related and is going to be impossible to fix and almost impossible to suppress, so we can really only disable LSan for those tests.

m-chaturvedi, 2017-12-05:
There are five Valgrind issues as well:
```
==29788== Use of uninitialised value of size 8
==29788==    at 0xA750B3F: ???
==29788==    by 0x135C5DF7: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29788==    by 0x135C6158: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29788==    by 0x135C6833: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29788==    by 0x135C6656: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29788==    by 0xCE906B9: start_thread (pthread_create.c:333)
==29788==    by 0xD7463DC: clone (clone.S:109)
```
```
==29814== Use of uninitialised value of size 8
==29814==    at 0x4121D4C: ???
==29814==    by 0xF763ADC: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29814==    by 0xF764CD1: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29814==    by 0xF763158: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29814==    by 0xF763833: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29814==    by 0xF763656: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29814==    by 0xA0A56B9: start_thread (pthread_create.c:333)
==29814==    by 0xA95A3DC: clone (clone.S:109)
```
```
==29788== Conditional jump or move depends on uninitialised value(s)
==29788==    at 0xA7538C2: ???
==29788==    by 0x135C5DF7: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29788==    by 0x135C6158: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29788==    by 0x135C6833: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29788==    by 0x135C6656: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29788==    by 0xCE906B9: start_thread (pthread_create.c:333)
==29788==    by 0xD7463DC: clone (clone.S:109)
```

```
==29788== 112 (56 direct, 56 indirect) bytes in 1 blocks are definitely lost in loss record 1,674 of 1,731
==29788==    at 0x4C2FB55: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==29788==    by 0x13298C1B: ???
==29788==    by 0x13257F8B: ???
==29788==    by 0x13258078: ???
==29788==    by 0x132580E7: ???
==29788==    by 0x13258B11: ???
==29788==    by 0x1327A924: ???
==29788==    by 0x1327D330: ???
==29788==    by 0x132783E8: ???
==29788==    by 0x1327851F: ???
==29788==    by 0x1323C131: ???
==29788==    by 0x133268BE: ???
==29788==    by 0x1324290E: ???
==29788==    by 0x1323B6E9: ???
==29788==    by 0x1323BC78: ???
==29788==    by 0x135C4DB8: ???
```

```
==29814== Syscall param writev(vector[...]) points to uninitialised byte(s)
==29814==    at 0xA94FFBD: ??? (syscall-template.S:84)
==29814==    by 0xD7DFF28: ??? (in /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0)
==29814==    by 0xD7E031C: ??? (in /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0)
==29814==    by 0xD7E03A4: xcb_writev (in /usr/lib/x86_64-linux-gnu/libxcb.so.1.1.0)
==29814==    by 0xB3994AD: _XSend (in /usr/lib/x86_64-linux-gnu/libX11.so.6.3.0)
==29814==    by 0xB38E7AA: ??? (in /usr/lib/x86_64-linux-gnu/libX11.so.6.3.0)
==29814==    by 0xB38DA85: ??? (in /usr/lib/x86_64-linux-gnu/libX11.so.6.3.0)
==29814==    by 0xB38E9B4: XPutImage (in /usr/lib/x86_64-linux-gnu/libX11.so.6.3.0)
==29814==    by 0xB12CEE6: ??? (in /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0)
==29814==    by 0xB12CF12: ??? (in /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0)
==29814==    by 0xF3933A5: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29814==    by 0xF3938ED: ??? (in /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so)
==29814==    by 0xB12C924: ??? (in /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1.2.0)
==29814==    by 0x6DCFD6D: vtkRenderWindow::Render() (in /home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-memcheck-valgrind/_bazel_ubuntu/1b61c7568cad04c517ecba51f21eb5d9/external/vtk/lib/libvtkRenderingCore-8.0.so.1)
==29814==    by 0x6A0CBE0: vtkXOpenGLRenderWindow::Render() (in /home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-memcheck-valgrind/_bazel_ubuntu/1b61c7568cad04c517ecba51f21eb5d9/external/vtk/lib/libvtkRenderingOpenGL2-8.0.so.1)
==29814==    by 0x14876D: drake::systems::sensors::(anonymous namespace)::PerformVTKUpdate(vtkNew<vtkRenderWindow> const&, vtkNew<vtkWindowToImageFilter> const&, vtkNew<vtkImageExport> const&) (rgbd_renderer.cc:86)
```

jwnimmer-tri, 2018-11-14:
The best we can do is to suppress these warnings and/or on a case-by-case basis report them upstream to VTK if there's anything to be done on that side.  We don't need this tracking issue.

PR #7541 (2017-11-30): Fix uninitialized value in rgbd_camera.cc
Came up when I was trying to fix #7520 on running with Valgrind.

This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7541)
closed 2018-11-14 by jwnimmer-tri