RobotLocomotion/drake #7536: Support symbolic::Expression in MBT code

amcastro-tri, 2017-11-29:
The first step towards supporting symbolic types in MBT code is making `RotationInertia` compile with `symbolic::Expression`.

In particular, @soonho-tri, how would I need to write the method `RotationalInertia<T>::AreMomentsOfInertiaNearPositive` to support symbolic types?

My dev branch [`symbolic_rotational_inertia`](https://github.com/amcastro-tri/drake/tree/symbolic_rotational_inertia) specializes in `symbolic::Expression` those methods that do not even compile and provides a dummy default implementation that only aborts, see [`rotational_inertia.cc`](https://github.com/amcastro-tri/drake/blob/symbolic_rotational_inertia/drake/multibody/multibody_tree/rotational_inertia.cc). Those are the first things to resolve.

soonho-tri, 2017-11-30:
FTR, DM-ed in Slack.

amcastro-tri, 2017-12-11:
The soon to be merged `RotationMatrix` in #7488 is probably a smaller class to play with first. Same problems; methods with the flavor of a "CheckInvariants(tolerance)" which do work well with `double` do not usually work with expressions (typically because they return `bool`). Also, these methods are many times used within a `DRAKE_ASSERT_VOID" or similar to check for invariants in Debug builds only.

soonho-tri, 2017-12-11:
First of all, we need to change the return type of `AreMomentsOfInertiaNearPositive` from `bool` to `decltype(T() < T())`. Note that `decltype(T() < T())` is `bool` when `T = double` while `decltype(T() < T())` is `symbolic::Formula` when `T = symbolic::Expression`. I think with this change, we can make `AreMomentsOfInertiaNearPositive` compatible with symoblic.

jwnimmer-tri, 2019-03-04:
Fixed by #10535 and #10439.

closed 2019-03-04 by jwnimmer-tri