RobotLocomotion/drake #7558: Input not connected error is unhelpful

ggould-tri, 2017-12-04:
I accidentally failed to wire up an input.  Here is the error I got:
```
abort: Failure at external/drake/drake/systems/framework/system.h:407 in EvalEigenVectorInput(): condition 'input_vector != nullptr' failed.

*** 1 failure is detected in the test module "Master Test Suite"
Running 5 test cases...
unknown location(0): fatal error: in "TrajectoryTesterSmokeTest": signal: SIGABRT (application abort requested)
program/controller_analyzer/test/run_one_trajectory.cc(46): last checkpoint
ERROR: Non-zero return code '201' from command: Process exited with status 201
```
This of course does not actually indicate *which* system isn't wired.  Doubly annoyingly, the system in question doesn't even appear in the stack if you gdb it; static inspection is the only thing that works.

liangfok, 2017-12-04:
Does the graphviz visualization of a diagram help in this case: #5545

ggould-tri, 2017-12-04:
It would probably be helpful, but seems like overkill -- the error message could just say where the error was.  Having the user indirect through graphviz for information the diagrambuilder should already have is a bit out-of-the-way (and surprising for a novice user -- the Graphviz thing is only mentioned in one place in the documentation, and not linked to anywhere that anyone would get to from this error message).

ggould-tri, 2017-12-04:
(note that doxygen search doesn't search the text of descriptions or substrings of method names, so good luck finding the graphviz support from the docs!)

ggould-tri, 2017-12-04:
(Filed #7559 for that documentation issue)

jwnimmer-tri, 2018-01-25:
Even worse, with assertions off you actually get a segfault, not even the assertion text!

PR #7852 (2018-01-25): Raise an error for EvalEigenVectorInput on a disconnted input
By default, we segfault (release) or assert (debug).  These are not helpful for users.

Fixes #7558.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7852)
closed 2018-01-25 by jwnimmer-tri