RobotLocomotion/drake #7559: Drake docs appear searchable but really aren't

ggould-tri, 2017-12-04:
Liang helpfully pointed me to Drake's graphviz support today; this was especially helpful because it is nearly impossible to discover that Drake has graphviz support.

For instance, if you search Drake doxygen for "graphviz" it finds nothing.
[image]
If you search Drake's top-level docs it finds nothing
[image]
In fact if you search google it finds nothing
[image]

A similar situation applies to other Drake concepts that are not exact class names; for example "algebraic loop" or "inverse kinematics".

The underlying issue is probably in part that (1) the search at drake.mit.edu does not reach the doxygen, and (2) the doxygen search only matches exact class and method names, not doxygen comment content or words appear in class or method names.

jamiesnape, 2017-12-14:
Googling `graphviz site:drake.mit.edu` gives me around 100 results.

ggould-tri, 2017-12-14:
Good point.  We should perhaps put a google custom search on the drake.mit.edu page.  cse.google.com generates custom search boxes for searches of that type, although iirc they have to be owned by a particular google account so we should choose carefully who creates it.

RussTedrake, 2018-01-31:
if it makes sense for me to volunteer my google account for this, i'm happy to do it.

SeanCurtis-TRI, 2018-01-31:
Can't we simply make an account for drake? And then we can use LastPass to share the drake password as necessary.

SeanCurtis-TRI, 2020-05-14:
@jwnimmer-tri Given the team and component, shall we reassign this?

jwnimmer-tri, 2020-05-14:
If you're no longer interested in working on it, then I'm content for you to assign it to me.

RussTedrake, 2020-06-26:
I’d like to upvote the priority for this.  The search is really a limitation, and gives the impression to new users that we don’t have good documentation (even though we do!).
  
Before rediscovering this thread, I was looking at Doxygen’s search offerings: https://www.doxygen.nl/manual/searching.html
I don’t know that we’ve tried the trivial experiments.  But we definitely don’t have any of them enabled in master: https://github.com/RobotLocomotion/drake/blob/master/doc/Doxyfile_CXX.in#L222

But I agree that going with a proper search engine like google would be far better.

FTR — I also compiled myself a docset from master and am trying out [Dash](https://kapeli.com/dash).  My first impression is pretty positive — it’s nice to search the docs for drake, Eigen, matplotlib, etc, all in one place.

jamiesnape, 2020-06-26:
If you don't mind me using one of our bot's Google Accounts, I can set up Custom Search easily enough.

RussTedrake, 2020-06-27:
I can't think of any reason that would be a problem.

jamiesnape, 2020-07-01:
You can test the search from here: https://cse.google.com/cse?cx=000573978146867327766:lutxsxsejrg. I will need to embed some HTML into the various pages next.

RussTedrake, 2020-07-01:
Thanks Jamie.  Poking around at it, the results for "urdf" and "graphviz" weren't quite as helpful as I'd hoped...

jwnimmer-tri, 2021-03-15:
In the developers call today, it was noted that DDG site search may also be an option.

EricCousineau-TRI, 2021-03-15:
See examples also:

* https://www.google.com/search?&q=site%3Adrake.mit.edu+pydrake&oq=site%3Adrake.mit.edu+pydrake+install
* https://duckduckgo.com/?q=site%3Adrake.mit.edu+pydrake+install

Came up in Drake Slack: https://drakedevelopers.slack.com/archives/C01879RNKQW/p1615817907049500

jwnimmer-tri, 2022-01-31:
See #16473 for discussion of adding DDG search to the `pydrake` docs.  I also have a branch for adjusting the Doxygen search box, but it isn't quite ready yet.

jwnimmer-tri, 2022-02-01:
The new pydrake dual-search is live now: https://drake.mit.edu/pydrake/.

The Doxygen dual-search is in progress with #16482 and the one more PR after that.

jwnimmer-tri, 2022-02-02:
I've forked off the Jekyll searchbox into a separate ticket (#16482).

That just leaves the Doxygen fix as part of this issue.  (I don't think we need to add a searchbox to the style guide.)

jwnimmer-tri, 2022-02-02:
The Doxygen search is finally ready for commentary now, at #16491.

PR #16491 (2022-02-02): [doc] Add DDG searches to our Doxygen header
Closes #7559.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/16491)
closed 2022-02-02 by jwnimmer-tri