RobotLocomotion/drake #7562: Systems framework should fail-fast if feedthrough declarations are disobeyed at runtime

jwnimmer-tri, 2017-12-05:
If a System declares that output N doesn't use input M, but then at runtime _does_ access input M when computing output N, it would help if the framework produced an exception, instead of a (potential) stack overflow.

Given the current APIs, we suspect this is hard to enforce in general (perhaps until caching lands), but for the 90% case of either _all_ or _no_ feedthrough allowed, we think we should be able to fail-fast.

Relatedly, we could also fail-fast on stack overflows (`mutable bool is_under_evaluation;` in an OutputPort).

jwnimmer-tri, 2018-10-28:
Over to @sherm1 for vote / close / backlog prioritization.

sherm1, 2020-05-11:
#12709 is a modernized version of this issue; closing the older one.

closed 2020-05-11 by sherm1