RobotLocomotion/drake #7583: pydrake runtime error: Mosek is not installed in your build. You'll need to use a different solver.

weiqiao, 2017-12-11:
I'm calling Mosek solver in python as in drake/drake/bindings/pydrake/test/testMosekSolver.py
Then line 
```
result = solver.Solve(prog)
```
triggers the runtime error
```
RuntimeError: Mosek is not installed in your build. You'll need to use a different solver.
```

I checked that Mosek works in C++ interface.

weiqiao, 2017-12-11:
cc @hongkai-dai

weiqiao, 2017-12-11:
cc @jwnimmer-tri

soonho-tri, 2017-12-13:
@weiqiao , what do you have when you run `bazel build //bindings/pydrake/solvers/... --config=mosek && ./bazel-bin/bindings/pydrake/solvers/mosek_solver_test`?

weiqiao, 2017-12-13:
I got error 
```
ERROR: Skipping '//bindings/pydrake/solvers/...': no targets found beneath 'bindings/pydrake/solvers'

```

soonho-tri, 2017-12-13:
Recently we removed `drake` directory and moved every sub-directory one-level up. Can you check-out the latest version and re-test it?

jamiesnape, 2017-12-13:
Yes, we probably need to fix up the RPATH on the MOSEK library that we download (and again if we copied it to the install prefix). GUROBI is a little more awkward as it is not in the sandbox. We could fix up the RPATH, but it might be seen as impolite to modify something outside of the sandbox. There would be no issues on Mac, though.

jamiesnape, 2017-12-13:
Actually ignore that. We would be modified the Drake shared objects, not Gurobi, so it would be fine.

jamiesnape, 2017-12-13:
Opened #7592 which would remove the `Experimental` notation no the `WITH_GUROBI` and `WITH_MOSEK` options in CMake.

EricCousineau-TRI, 2018-01-28:
Should be addressed by #7598 and #7873.

closed 2018-01-28 by EricCousineau-TRI