RobotLocomotion/drake #7586: Removal of top-level drake-distro broke CLion

sherm1, 2017-12-12:
Per discussion on Slack #clion channel, the rename of all files from `drake/blah` to just `blah` had some unforseen consequences. Our header includes look like `drake/common/something.h` which could be resolved by starting at the top-level directory (`drake-distro` or whatever). Now there is no such file name so we are crafting symbolic links in `~/.cache/bazel` in a mock top-level directory.

CLion is not amused by this. When following symbols, instead of opening header files under their original names it opens the (very long named) symbolic links, which are read only and not recognized as being part of the project. Edits made to the correctly-named files do not trigger re-indexing so the new symbols are not recognized until a manual Bazel Sync is done (a slow process). It is not clear whether other source-aware IDEs or tools would have similar confusion.

Other projects deal with this issue in various ways. However, a popular one is to introduce an extra level in the git repo, for example @jwnimmer-tri found [tensorflow/tensorflow](https://github.com/tensorflow/tensorflow). That is what we were doing before -- maybe it was fine as it was?

sherm1, 2017-12-12:
/cc @RussTedrake @jamiesnape @stonier @SeanCurtis-TRI @EricCousineau-TRI

sherm1, 2017-12-12:
Possibly related: bazelbuild/intellij#130

jwnimmer-tri, 2017-12-13:
For the moment, we have a work-around:

Assuming that you have Drake checked out as `/home/foo/stuff/drake` (that is the directory `/home/foo/stuff/drake/.git` exists) then you may hack the Bazel plug-in's source code (in your home directory) as follows:

```console
~$ git diff .CLion2017.2/config/plugins/clwb/aspect/intellij_info_impl.bzl{.orig,}
diff --git a/.CLion2017.2/config/plugins/clwb/aspect/intellij_info_impl.bzl.orig b/.CLion2017.2/config/plugins/clwb/aspect/intellij_info_impl.bzl
index d2be0fa..e56d685 100644
--- a/.CLion2017.2/config/plugins/clwb/aspect/intellij_info_impl.bzl.orig
+++ b/.CLion2017.2/config/plugins/clwb/aspect/intellij_info_impl.bzl
@@ -349,7 +349,7 @@ def collect_cpp_info(target, ctx, ide_info, ide_info_file, output_groups):
       target_define = target_defines,
       target_copt = target_copts,
       transitive_include_directory = cc_provider.include_directories,
-      transitive_quote_include_directory = cc_provider.quote_include_directories,
+      transitive_quote_include_directory = ["/home/foo/stuff"] + cc_provider.quote_include_directories,
       transitive_define = cc_provider.defines,
       transitive_system_include_directory = cc_provider.system_include_directories,
   )
```

In other words, you can tell CLion that the folder that _contains_ Drake should be on your include path; then, CLion resolves include statements into the source code under git, instead of the build system's symlinks to it.  This does _not_ change how the code is compiled by Bazel, only how the include graph is perceived by CLion.

sherm1, 2017-12-13:
(Notes from #clion slack discussion)
This patch helps a lot, with some limitations:
- all of your top-level repo directories must be named `drake` (e.g. `drake-distro1/drake`, `drake-distro2/drake`). Making `drake` a symbolic link causes CLion trouble.
- clicking on compilation Error messages that occurred in headers still takes you to the _virtual_includes path. Editing is allowed there but seems to confuse CLion; I lost some edits and had to put them back into the file under the original name.

sherm1, 2018-01-02:
I'm running into trouble with this workaround now. I'm attempting to use a second cloned repo "drake2/drake" to extract some code changes from my mega-caching branch in "drake" to more tractable smaller branches. Both projects are up in CLion at the same time for convenient cut-and-paste operation, and I want to be able to do builds in both. But when I select symbols in the "drake2/drake" project, it pulls up files in the "drake" repo instead. It does that quietly and the files have the right names -- but they are the wrong files! 

This doesn't seem workable for me -- I don't think I can remain sufficiently vigilant to avoid disaster. Is there a better way?

SeanCurtis-TRI, 2018-01-03:
@jwnimmer-tri I'm not familiar with skylark; but is it possible to get the repo root from the `target` or `ctx` variable that is passed into `collect_cpp_info` and pre-pend the parent directory to the list? That might make it adapt to the repo for each session of CLion.

jwnimmer-tri, 2018-01-05:
@SeanCurtis-TRI When I first created the patch, I looked into doing that but didn't obviously see a way.  (In general, skylark code isn't really supposed to know about the whole filesystem, or which workspace its in, or etc.)  When I have some time, I will take another look (for either that kind of tix, or other change) and post back here, but it might not be particularly soon.

One brutal but probably effective work-around is to install multiple copies of CLion (one copy of CLion per Drake checkout).  I haven't tried that, but from what I recall of the installation, it lets you choose the install folder, scratch folder, etc. at will.

avalenzu, 2018-01-23:
Is this still an open question? Or have we decided to just stick it out with the current (post-hoist) directory layout? If we are still considering a reversion, I'd like to note that the old structure (`drake/drake`) made full-text searching in our code-base much simpler (since you could just search in `drake/drake` without having to ignore the `bazel-*` directories).

jwnimmer-tri, 2018-01-23:
Its on my near-term TODO list to revisit the CLion question.

For searching, my emacs built-in is `find . -type f -print0 | xargs -0 grep -n -e needle`, which by default does not traverse into symlinks.  Also `git grep needle` works well.

avalenzu, 2018-01-23:
Just checked the `ack` command I was using, and the symlinks were a red-herring (`ack` was excluding them as well); turns out I had an old `build` directory around that was slowing down the search. Sorry for the noise.

jwnimmer-tri, 2018-01-26:
> For the moment, we have a work-around: ... you may hack the Bazel plug-in's source code (in your home directory) ...

Note to anyone who hacked the CLion plug-in source as the work-around.  You should undo the hack, and instead follow the new instructions on http://drake.mit.edu/clion.html that relate to `tools/clion/bazel_wrapper`.

jwnimmer-tri, 2018-01-26:
> ... clicking on compilation Error messages that occurred in headers still takes you to the _virtual_includes path. Editing is allowed there but seems to confuse CLion; I lost some edits and had to put them back into the file under the original name.

I have a prototype fix for this symptom as well at https://github.com/jwnimmer-tri/drake/tree/clion2.  I plan to roll it out after folks pull the first fix (already merged as #7843) and see how it works.

PR #8016 (2018-02-28): Update to Bazel 0.10.1 on Ubuntu and improve CLion integration
See commit messages for details.

Closes #7586.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8016)
closed 2018-02-28 by jwnimmer-tri