RobotLocomotion/drake #7587: Build failure in lcmtypes due to use of system python-protobuf

avalenzu, 2017-12-12:
This appears to be unique to my machine, which is rather troubling given the reputability we expect from Bazel. If `python-protobuf` is installed, I get the following output

```
$ bazel build //lcmtypes:lcmtypes_py 
INFO: Analysed target //lcmtypes:lcmtypes_py (0 packages loaded).
INFO: Found 1 target...
ERROR: <REDACTED>/drake-distro/lcmtypes/BUILD.bazel:237:1: SkylarkAction lcmtypes/lcmt_bicycle_car_state_t.lcm failed (Exit 1)
/usr/lib/python2.7/dist-packages/google/protobuf/descriptor.pyc # ADDED LINE
Traceback (most recent call last):
  File "<REDACTED>/.cache/bazel/_bazel_<REDACTED>/983b846db67aead8f755afbe4b46c3ce/bazel-sandbox/3356917300376060265/execroot/drake/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/../drake/tools/vector_gen/lcm_vector_gen.py", line 15, in <module>
    from drake.tools.vector_gen import named_vector_pb2
  File "<REDACTED>/.cache/bazel/_bazel_<REDACTED>/983b846db67aead8f755afbe4b46c3ce/bazel-sandbox/3356917300376060265/execroot/drake/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/tools/vector_gen/named_vector_pb2.py", line 22, in <module>
    serialized_pb=_b('\n#tools/vector_gen/named_vector.proto\x12\x05tools\"L\n\x0bNamedVector\x12*\n\x07\x65lement\x18\x01 \x03(\x0b\x32\x19.tools.NamedVectorElement\x12\x11\n\tnamespace\x18\x02 \x01(\t\"\x7f\n\x12NamedVectorElement\x12\x0c\n\x04name\x18\x01 \x01(\t\x12\x0b\n\x03\x64oc\x18\x02 \x01(\t\x12\x15\n\rdefault_value\x18\x03 \x01(\t\x12\x11\n\tdoc_units\x18\x04 \x01(\t\x12\x11\n\tmin_value\x18\x05 \x01(\t\x12\x11\n\tmax_value\x18\x06 \x01(\t')
TypeError: __init__() got an unexpected keyword argument 'syntax'
Target //lcmtypes:lcmtypes_py failed to build
Use --verbose_failures to see the command lines of failed build steps.
INFO: Elapsed time: 0.643s, Critical Path: 0.21s
FAILED: Build did NOT complete successfully
``` 
whereas if I `apt remove` the `python-protobuf` package, `//lcmtypes:lcmtypes_py` builds successfully. 

The line marked `ADDED LINE` in the output above was generated by adding the following lines after line 11 of tools/vector_gen/lcm_vector_gen.py:
```python
import sys
from google.protobuf import descriptor as _descriptor
sys.stderr.write(_descriptor.__file__ + "\n")
```
That line indicates that Bazel is getting `descriptor` from my system `python-protobuf` rather than the one built in Drake. @EricCousineau-TRI has some speculation on why that might be the case - I'll let him add that in the comments.

I can work around this by uninstalling `python-protobuf`, but it may be indicative of an upstream bug in Bazel's handling of protobuf targets.

cc. @jamiesnape, @jwnimmer-tri

jwnimmer-tri, 2017-12-12:
As of #7323 we are going to use system protobuf on purpose, so this symptom at least will not be a problem.  Possibly other third-party workspace python items could be, though I think those are relatively few.

EricCousineau-TRI, 2017-12-12:
We had initially thought that `drake_py_proto_library` had somehow missed the Bazel `protobuf`s Python runtime environment, but inspecting `@com_google_protobuf//:protobuf.bzl` shows that it should be there by default, so I'm not sure at this point.

Not exactly sure why this happens on Andres's system, as it works fine on my system, even with `python-protobuf` installed.
From my desktop:
```
$ apt search protobuf | grep python
python-protobuf/xenial,now 2.6.1-1.3 amd64 [installed]
python-protobuf.socketrpc/xenial,xenial 1.3.2-3 all
```
I believe Andres had his installed from a PPA.

I am not sure how this will translate to other Python libraries, even after #7323; hopefully this is isolated to some sort of magic specific to this PPA version of `protobuf`?

avalenzu, 2017-12-12:
While I originally had `python-protobuf` installed from a PPA, the problem persisted after removing the PPA apt purging `python-protobuf` and re-installing it from the default repositories.

jwnimmer-tri, 2018-02-03:
Given that we're on system protobuf now on purpose, I think we can close and then any further problems with other Python packages we can investigate as they occur.  (Please re-open if I'm mistaken.)

closed 2018-02-03 by jwnimmer-tri