RobotLocomotion/drake #7588: Add example for simulation with multi-process communication

EricCousineau-TRI, 2017-12-12:
Per conversation during Drake Developers meeting, @dmsj requested an example showing how to do a simulation with multi-process communication.

@avalenzu mentioned that he has a PR that should land soon that would encapsulate this.

Some of the potential options:
* `LcmDrivenLoop` - what Andres has
* Add in sleep / synchronization / blocking mechanisms (e.g. ROS service calls)
* Implement a simple clock interface, to generalize `LcmDrivenLoop` (to ease the creation of something like `RosDrivenLoop`) and synchronize with `Simulator::set_target_realtime_rate`

\cc @jwnimmer-tri @avalenzu @sammy-tri @siyuanfeng-tri

jwnimmer-tri, 2019-07-02:
I am not sure that we plan to work on this at all.  Maybe we can close it?  @avalenzu @EricCousineau-TRI?

avalenzu, 2019-07-08:
This was resolved by #7610. However, that example went away with the removal of all pick-and-place code in #10591. Since I should have closed this issue when #7610 merged, I'm going to close it anyway. If anyone feels this is a current pain point, he or she can feel free to re-open it.

closed 2019-07-08 by avalenzu