RobotLocomotion/drake #7593: FindResource no longer works when running binaries directly

avalenzu, 2017-12-13:
`FindResource()` and friends appear not to be happy about the hoist:

Before #7575 
```
$ git checkout 7f23076
$ bazel run //drake/examples/kuka_iiwa_arm:iiwa_controller
<snip>
INFO: Running command line: bazel-bin/drake/examples/kuka_iiwa_arm/iiwa_controller
[2017-12-13 10:54:31.693] [console] [info] Welding joint iiwa_joint_ee
[2017-12-13 10:54:31.693] [console] [info] Welding joint tool0_joint
[2017-12-13 10:54:31.700] [console] [info] controller started
^C
Bazel caught interrupt signal; shutting down.
$ bazel-bin/drake/examples/kuka_iiwa_arm/iiwa_controller
[2017-12-13 10:55:08.554] [console] [info] Welding joint iiwa_joint_ee
[2017-12-13 10:55:08.555] [console] [info] Welding joint tool0_joint
[2017-12-13 10:55:08.558] [console] [info] controller started
^C
```

After #7575 
```
$ git checkout c84bceb
$ bazel run //examples/kuka_iiwa_arm:iiwa_controller
<snip>
INFO: Running command line: bazel-bin/examples/kuka_iiwa_arm/iiwa_controller
[2017-12-13 10:50:42.587] [console] [info] Welding joint iiwa_joint_ee
[2017-12-13 10:50:42.587] [console] [info] Welding joint tool0_joint
[2017-12-13 10:50:42.595] [console] [info] controller started
^C
Bazel caught interrupt signal; shutting down.
$ bazel-bin/examples/kuka_iiwa_arm/iiwa_controller
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not find resource: drake/manipulation/models/iiwa_description/urdf/iiwa14_polytope_collision.urdf
[1]    2175 abort (core dumped)  bazel-bin/examples/kuka_iiwa_arm/iiwa_controller

```

avalenzu, 2017-12-13:
Apparently this issue only appears if your Drake checkout is named something other than `drake`. `drake-distro` was a common name for such checkouts in the past. Renaming the checkout is a workaround, but ultimately, `FindResource()` shouldn't care what your checkout is named.

rdeits, 2017-12-14:
That seems like a somewhat unfriendly requirement, especially for those of us with lots of copies of drake floating around.

jwnimmer-tri, 2017-12-14:
Yup, this is a regression and we'll fix it soon.

PR #7614 (2017-12-15): Allow for Drake checkouts not named drake
When searching from `${PWD}` up the directory tree to find Drake's resources (during development, not install) allow for the top-level folder not to be named "drake".

Fixes #7593.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7614)
closed 2017-12-15 by jwnimmer-tri