RobotLocomotion/drake #7609: MacOS setup missing brew link step

ahundt, 2017-12-14:
I think ipopt is keg only so this might need to be added to the setup script:

```brew link ipopt```

soonho-tri, 2017-12-14:
> I think ipopt is keg only

I don't think `ipopt` is keg-only. Could you check https://github.com/Homebrew/homebrew-science/blob/master/ipopt.rb ?

jamiesnape, 2017-12-14:
If it were keg only, we definitely should not do this; things are keg only for reason. However, ipopt is not keg only.

soonho-tri, 2017-12-14:
I think we can close this issue. @ahundt , please feel free to add a comment or re-open it if you need.

closed 2017-12-14 by soonho-tri

ahundt, 2017-12-14:
ok no worries, probably just got unlinked on my machine for some reason or another. Sorry for the flurry of issues this is my first time installing + building.