RobotLocomotion/drake #7615: bazel downgraded during install on 16.04

ahundt, 2017-12-14:
I am going through a first time installation on 16.04 and while I was aware that the instructions noted the setup script would not be conservative, it seems my bazel version was downgraded from 0.7.x to 0.6.x, which I didn't expect. Two questions:

1. Can this be reversed/re-upgraded, ideally without breaking drake?
2. Are there many other packages where there might be major downgrades?  

Thanks!

Output below:
```

--2017-12-14 18:20:23--  https://github.com/bazelbuild/bazel/releases/download/0.6.1/bazel_0.6.1-linux-x86_64.deb
Resolving github.com (github.com)... 192.30.253.113, 192.30.253.112
Connecting to github.com (github.com)|192.30.253.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/20773773/31af8dac-a9de-11e7-96da-66bd626f63e7?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20171214%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20171214T232023Z&X-Amz-Expires=300&X-Amz-Signature=479fa92c23810c1de18fa5df8b343626ecb639a41f43954ae2ccbdef8f0123fe&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dbazel_0.6.1-linux-x86_64.deb&response-content-type=application%2Foctet-stream [following]
--2017-12-14 18:20:23--  https://github-production-release-asset-2e65be.s3.amazonaws.com/20773773/31af8dac-a9de-11e7-96da-66bd626f63e7?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20171214%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20171214T232023Z&X-Amz-Expires=300&X-Amz-Signature=479fa92c23810c1de18fa5df8b343626ecb639a41f43954ae2ccbdef8f0123fe&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dbazel_0.6.1-linux-x86_64.deb&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)... 52.216.20.131
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.20.131|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 92268204 (88M) [application/octet-stream]
Saving to: ‘/tmp/bazel_0.6.1-linux-x86_64.deb’

/tmp/bazel_0.6.1-linux-x 100%[================================>]  87.99M  59.4MB/s    in 1.5s

2017-12-14 18:20:24 (59.4 MB/s) - ‘/tmp/bazel_0.6.1-linux-x86_64.deb’ saved [92268204/92268204]

/tmp/bazel_0.6.1-linux-x86_64.deb: OK
dpkg: warning: downgrading bazel from 0.8.1 to 0.6.1
(Reading database ... 732151 files and directories currently installed.)
Preparing to unpack .../bazel_0.6.1-linux-x86_64.deb ...
Unpacking bazel (0.6.1) over (0.8.1) ...
Setting up bazel (0.6.1) ...
+ bazel build //...
Extracting Bazel installation...
```

ahundt, 2017-12-14:
The build also seems to have been stuck at this point for around an hour, it is a fast machine with a good connection so I think there may be a problem. Please advise?

```

Setting up bazel (0.6.1) ...
+ bazel build //...
Extracting Bazel installation...
...............
Analyzing: 4695 targets (211 packages loaded)
    currently loading: @eigen//
    Fetching https://drake-packages.csail.mit.edu/...-vtk-8.0.1-xenial-x86_64.tar.gz; 6,044,848b
    Fetching https://drake-packages.csail.mit.edu/...1-qt-5.5.1-xenial-x86_64.tar.gz; 1,580,234b
    Fetching https://github.com/bulletphysics/bullet3/archive/2.86.1.tar.gz; 2,313,716b
    Fetching https://github.com/.../fcl/archive/06d48b3b6f3605b8caf119d5208d8156eb64fe0d.tar.gz
    Fetching https://github.com/bazelbuild/buildtools/releases/download/0.6.0/buildifier

```

jwnimmer-tri, 2017-12-15:
> Can [the bazel downgrade] be reversed/re-upgraded, ideally without breaking drake?

Probably yes.  Many of us use the newest version locally, even though Drake's CI-supported version (which is what the script installs) tends to lag a bit.  I know at least 0.8.0 and I think 0.8.1 should work.

We should probably change out setup script to warn / confirm with the user on downgrade, though.

> Are there many other packages where there might be major downgrades?

Probably not.  The `bazel` package is the only one we install via `dpkg`; the rest we just let `apt` handle (with one extra non-Ubuntu source entry for clang 3.9).

The details are in https://github.com/RobotLocomotion/drake/blob/master/setup/ubuntu/16.04/install_prereqs.sh; its relatively easy to skim, if you want to get the general idea.

> The build also seems to have been stuck at this point for around an hour ...

I don't have a great idea on this one; some of the downloads _are_ relatively large.  Unfortunately the progress report on downloads aren't superb.

Actually one idea -- did you perhaps turn on snopt, but not set up git authentication (so its stuck at a command prompt)?  (Look at `ps auxf` or similar to see what child processes bazel has.)

ahundt, 2017-12-15:
Hmm so this install script did something very destructive. I haven't taken any steps to diagnose it yet but I cannot log into my computer from the Ubuntu GUI. I'll see if I can get in via terminal...

Do you have any advice on what to check?

ahundt, 2017-12-15:
what are ibex and dreal?

jwnimmer-tri, 2017-12-15:
Other than downgrading bazel, the script just installs a number of new packages from Ubuntu itself.  Possibly some other upgrades happened.  Does a reboot help?  Do you have NVidia or GL drivers that would have gotten mixed up concurrently?

If you ran yesterday, IBEX and dReal would not have come in -- those were must merged into master moments ago.  In any case, they are just some shared libraries and a helper in `/usr/bin`, I would be surprised if they were relevant.

soonho-tri, 2017-12-15:
dReal is an SMT solver and IBEX is a dependency of dReal. Note that they are just added to the `install_prereqs.sh` script and not used yet. So it should not cause any problem for now.

ahundt, 2017-12-15:
Yes, I already had the nvidia ppas set up and I use them very frequently via tensorflow. That seems like one likely culprit

ahundt, 2017-12-15:
Hmm but my filesystem (btrfs) is read only at the moment. Note I didn't have anything else running at the time I installed things.

I ran this script which I wrote based on the install instructions:
https://github.com/ahundt/robotics_setup/blob/master/locomotion_robotics.sh

There are two changes I can't commit right now due to the read only problem. I removed the pip install line and I corrected `pip install --user --upgrade numpy lxml scipy yaml` to `pip install --user --upgrade numpy lxml scipy pyyaml`.

jwnimmer-tri, 2017-12-15:
Usually on Ubuntu, we would not be running anything related to pip.  Though I can't see how that would lead to a boot / login problem.

jwnimmer-tri, 2017-12-15:
You could look in `/var/log/dpkg.log` or `/var/log/apt/*.log` to see if something got removed due to a conflict resolution, perhaps?  Or in general for what happened.

ahundt, 2017-12-15:
Fortunately, rebooting a second time got me back! I'm a little worried about re-trying the script haha... I should at least turn off the bazel steps for certain. However the problem happened after I ran `bazel build //...`

> Usually on Ubuntu, we would not be running anything related to pip. Though I can't see how that would lead to a boot / login problem.

The instructions for spartan are also in that script since I'm trying out the kuka drivers. Those instructions require scipy and pyyaml from apt but in my environment I already have them installed via pip and I depend on those for my projects. In this case I figured such a minor change should be OK.

ahundt, 2017-12-15:
What should I look for in the log?

ahundt, 2017-12-15:
I created a pull request to default to the installed bazel:
https://github.com/RobotLocomotion/drake/pull/7627

ahundt, 2017-12-15:
@jwnimmer-tri @soonho-tri How would you guys feel about using the google provided apt repository for bazel by default?

They don't have a full API stability guarantee until 1.0 is out, but they do say they will "try very hard" to keep it stable between now an 1.0. If that issue means you can't do it now, perhaps once 1.0 is released?
https://bazel.build/support.html

Here's a script to install:
```
echo "deb [arch=amd64] http://storage.googleapis.com/bazel-apt stable jdk1.8" | sudo tee /etc/apt/sources.list.d/bazel.list
curl https://storage.googleapis.com/bazel-apt/doc/apt-key.pub.gpg | sudo apt-key add -
sudo apt-get update
sudo apt-get install -y bazel
sudo apt-get upgrade bazel
```

This is my version of it:
https://github.com/ahundt/robotics_setup/blob/master/bazel.sh

soonho-tri, 2017-12-15:
An issue is backward-compatibility. We have experienced problems as we bumped our 'official' bazel version and had to update Drake's `.bzl` files to be compatible with newer versions. `install_prereqs.sh` is our way of saying "At least, it works with those packages (+ specific version information)".

jwnimmer-tri, 2017-12-15:
We can see about once 1.x is released.  For now, we value (mostly) repeatable builds.  For a given drake commit, we know that it passed CI with a locked-down Bazel version as reflected on the source tree.  Using a PPA means that the version would be floating, so we could not easily build older Drake commits successfully anymore.  In any case, we will be updating to 0.8.1 soon, but need to test our developers' IDE integration before upgrading, which is a manual process.

ahundt, 2017-12-15:
> We can see about once 1.x is released. For now, we value (mostly) repeatable builds. 

That trade off makes sense, thanks for the details!

jwnimmer-tri, 2017-12-16:
Did anything come out of the "stuck downloading" problem?  I guess it resolved on its own?  With the Bazel downgrade made optional, I think both original topics are closed(ish).

I will kick it over to @sammy-tri for how to get an IIWA demo up and running, that's not my area of expertise (yet).

ahundt, 2017-12-16:
Ok thanks! I consider this closed then and will continue from here in RobotLocomotion/drake-iiwa-driver#15

closed 2017-12-16 by ahundt