RobotLocomotion/drake #7624: pydrake: Installed version can only use `GetDrakePath` once before it fails?

EricCousineau-TRI, 2017-12-15:
When testing using the installed Python #7323, I encountered an issue where calling `GetDrakePath` more than once causes it to fail:
```
$ cd /home/eacousineau/proj/tri/repo/branches/drake-distro/master
$ mkdir -p build_install
$ bazel --bazelrc=/dev/null run //:install -- ~+/build_install
$ export PYTHONPATH=~+/build_install/lib/python2.7/site-packages:${PYTHONPATH}
$ python -c 'import pydrake; print(pydrake.getDrakePath()); print(pydrake.getDrakePath())'
/home/eacousineau/proj/tri/repo/branches/drake-distro/master/build_install/share/drake/drake
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/eacousineau/proj/tri/repo/branches/drake-distro/master/build_install/lib/python2.7/site-packages/pydrake/__init__.py", line 14, in getDrakePath
    return abspath(common.GetDrakePath())
RuntimeError: could not find resource: drake/.drake-find_resource-sentinel
```

Possibly relates #7593?

\cc @jwnimmer-tri

jwnimmer-tri, 2017-12-15:
@fbudin69500 Would it make sense for you to look into this?

fbudin69500, 2018-01-08:
Sorry, I hadn't seen this issue until now. I'll look into it.

EricCousineau-TRI, 2018-01-08:
Awesome, thanks!

EricCousineau-TRI, 2018-01-09:
Closed via #7711.

closed 2018-01-09 by EricCousineau-TRI