RobotLocomotion/drake #7630: MathematicalProgram's symbolic constraint parser takes too long?

pangtao22, 2017-12-16:
I've been working on a quasistatic simulator (implemented as a drake discrete-time system) that solves an MIQP every time step to get the system states at the next time step. The specific system I am simulating consists of an Iiwa arm with a gripper and a rigid object on a table. In the simulation, the arm picks up the object and moves it to another location. 

I profiled my simulator on Mac using Instruments. But it seems that a significant amount of time (65.1%) was spent on `MathematicalProgram:AddLinearConstraint`. The profiling result (saved by Instruments) can be found below:
[4_sphere_example_quasistatic_system_profiling_result.trace.zip](https://github.com/RobotLocomotion/drake/files/1564442/4_sphere_example_quasistatic_system_profiling_result.trace.zip)

To reproduce this problem, 
1. Get my drake branch from https://github.com/pangtao22/drake/tree/quasistatic_3D_working/drake/manipulation/dev

2. Go to `drake-bistro/drake/manipulation/dev/`
3. bazel run --config=gurobi quasistatic_iiwa_pick_and_place_demo -- 12 1 0.01 // 12 means the system is simulated for 12 seconds, 1 is the BigM value given by the user (currently ignored), and 0.01 is the simulation time step.

pangtao22, 2017-12-16:
/cc @soonho-tri

RussTedrake, 2018-01-31:
@pangtao22 -- it's a lot to ask for setting up a mathematicalprogram to be performant enough to be done at every timestep.  have you looked at @siyuanfeng-tri's qp controller implementation?  he sets up the mathematicalprogram once, then only updates the constraints on each dt.

pangtao22, 2018-02-12:
At each time step, two `MathematicalProgram`s are solved:
1. an MIQP that calculates system configuration at the next time step.
2. a QP that minimizes the kinetic energy of the system.

Although it has more decision variables and constraints than the QP, the MIQP's size is constant, which  makes it ideal to construct the `MathematicalProgram` at the beginning of the simulation and only update the coefficients at each time step. In contrast, the smaller QP's size varies depending on the contact mode of the system. 

I implemented the MIQP in Siyuan's way and got a great speed improvement. For the iiwa pickup demo, `MathematicalProgram::AddLinearConstraint` now only takes 9.4% of the total time, as compared to 65.1% when a new `MathematicalProgram` was constructed at each time step.

closed 2018-07-03 by pangtao22