RobotLocomotion/drake #7656: FCL Integration Status

avalenzu, 2017-12-21:
This issue serves to track the status of our efforts to use FCL as the primary (only?) collision engine in Drake. The checklist below should be updated as tasks are completed.

FCL Upstream Fixes
- [ ] Submit FCL PR to fix inconsistent interpretation of capsule parameters (flexible-collision-library/fcl#225)
- [ ] Submit FCL PR to use special-case algorithm for Capsule-Capsule intersection (as it currently does for separation distance).
- [ ] Submit FCL PR to fix initialization of pointer members in `fcl::Convex` constructor.

libccd Upstream Fixes
- [ ] MPR collision point computation appears to be broken. [This branch](https://github.com/avalenzu/libccd/tree/mpr-penetration-fix) has what I (@avalenzu) believe is a fix.

There may be additional issues that we should consider in the discussion on ipab-slmc/exotica#184.

amcastro-tri, 2018-02-09:
@avalenzu. I was thinking about spiking an MBP/GS demo while at HQ in two weeks. Is the mesh (convex, just convex) penetration depth query working with FCL? would you have any unit test/example that shows failure modes?

sherm1, 2020-05-11:
@SeanCurtis-TRI is this worth keeping around?

SeanCurtis-TRI, 2020-05-12:
Yeah. We can close it. Of the remaining open issues, only the putative error in MRP collision point computation is of interest to me. The fix doesn't come with a test so it'll have to be investigated from scratch.

closed 2020-05-12 by SeanCurtis-TRI