RobotLocomotion/drake #7685: Clion refactor does not work after we move from drake-distro/drake to drake-distro

hongkai-dai, 2018-01-02:
Now when we try to refactor a function in Clion, it tries to modify a function in the .cache/bazel directory, instead of the actual code in drake-distro folder. Here is a screenshot when I try to rename a function in `mathematical_program.h`, it says I am trying to modify a non-project file.

[image]
[image]

cc @jwnimmer-tri

jwnimmer-tri, 2018-01-02:
Is this with or without the work-around in #7586 applied first?

hongkai-dai, 2018-01-02:
I didn't realize there is a work-around, let me try that.

closed 2018-01-05 by hongkai-dai