RobotLocomotion/drake #7702: add PCL as an external

RussTedrake, 2018-01-05:
@jamiesnape , @stonier, @jwnimmer-tri et al -- I'm am currently of the belief that we some code in drake master should be able to depend directly on PCL.  We've avoided this so far (I believe that @stonier is against it).  Can you help me consolidate the discussion here so that we make the decision?

Short motivation: drake is a library.  we've already enabled external projects (e.g. via shambhala) to use PCL, wihch is good.  But some of the components that we have authored there using PCL feel to me like core system building blocks that belong in the main library.  

I also believe that many subprojects will want to use drake and PCL.  Will having it linked directly make that easier or harder?

Related #7412

RussTedrake, 2018-01-05:
That doesn’t sound so bad?  There are already some pieces that are segregated (by not being included in libdrake.so) now, right?

One snag is that our rgb-d sim does use vtk right now, and it’s the perception pipeline that wants pcl.

jamiesnape, 2018-01-05:
> One snag is that our rgb-d sim does use vtk right now, and it’s the perception pipeline that wants pcl.

That is what I suspected. It all depends which parts of PCL they need, I guess.

EricCousineau-TRI, 2018-01-06:
Per Jamie's comment, I made a quick PR, RobotLocomotion/drake-shambhala#76, to narrow the number of PCL components. I also added in a quick no-op RGB-D camera simulation as well.
It shows that you cannot link `pcl_visualization` (`vtk-6.2`) and `drake` (`vtk-8.0`) into an application library.

@jamiesnape Are there any workarounds that you can think of for this? It would be nice to be able to quickly inspect things using PCL visualization.
I think I remember you had mentioned the possibility of having different versions of VTK linked into the same application; is there a way to magically sidestep these linker errors?

stonier, 2018-01-06:
**Linking Issues**

@jamiesnape / @sankhesh checked to make sure that the two different vtk's can be linked transitively - see the notes by sankesh [here](https://github.com/sankhesh/MultipleVTKVersions/blob/master/README.md). We went a step further and make sure vtk headers are private in drake so that someone is free to use system vtk 6.2 in their application (i.e. not just transitively) as well. OSRC checked against the ROS environment as well - however usage of pcl_visualizations there is restricted to external debugging applications only, so no need to worry about mixing it with Drake.

It obviously hasn't tickled the api extensively so there may be coverage issues with that. Your PR does not cause any problems when the visualization component is added (links fine for me), but I do get a segfault with an interesting backtrace - it travels from drake->vtk6.2->vtk8. What were your results?
We'll have to either fix those issues or consider stepping back to 6.2. Let's move this discussion to another thread?

**PCL in Drake**

I've got no problem with PCL in Drake per se. If I've given that impression inadvertently because it's been accidentally conflated with other problems (likely build and release work), my bad. If it's an enabler for point cloud based algorithms and it's no longer just a fringe experiment, then certainly.

There will be a build and release cost though. You should also keep in mind the version you choose to use for PCL pushes itself to users - it's a big, critical dependency they won't be free to be flexible with henceforth. If you go with a system version, as we have done with shambhala, you can be compatible with most people, but if you need to start running off a bleeding edge or forked PCL, then you'll have some harder decisions to make re compatibility.

RussTedrake, 2018-01-07:
thanks all.  I believe system pcl is aok.

jwnimmer-tri, 2018-01-17:
FYI I have PCL working in the Anzu repo, via Bazel, with no apparent problems.  Since that code doesn't use the RgbdCamera simulator, we don't even end up touching VTK8 so there's no possibility of trouble.  (Even if we mixed, I think it would still work given the linking research cited upthread.)

Porting the `@pcl_foo` targets from Anzu into Drake wouldn't be that hard, at least on Ubuntu.  We would have to see how macOS does.  And if we don't let PCL come into `libdrake.so` (i.e., only use it in `dev` code for now) then we don't need to solve the "infecting downstream with wrong-PCL" problem right away; we can experiment with PCL in Drake master but not ship it; then we'd even have some useful artifact to test against, when it came time to work on shipping PCL-related things to users.

RussTedrake, 2018-01-17:
SGTM

closed 2018-11-01 by RussTedrake