RobotLocomotion/drake #7725: URDF parser tries to load OBJ file instead of original STL file

christian-rauch, 2018-01-10:
I am trying to load a URDF file with mesh files names referring to STL files, e.g.:
```
<link name="base">
<visual>
    <geometry>
    <mesh filename="package://lwr_defs/meshes/lwr/base.stl"/>
    </geometry>
</visual>
</link>
```

The parser by itself removes the valid file extension `stl` and replaces it by `obj` and than failes to load the non-existent file: `RuntimeError: Unable to resolve an obj file from the filename "[...]/lwr_defs/meshes/lwr/base.obj" provided.`

How do I make sure that the URDF parser does not alter the filename by itself without permission?

jwnimmer-tri, 2018-01-10:
I'm not main developer for this piece, but have seen this as a user myself.  I believe #2941 is on point.  Drake's RigidBodyTree only supports `obj` for meshes at the moment; if an URDF specifies `.stl`, Drake will try to find an adjacent `.obj` file to load instead -- this allows the URDF to remain unchanged, but a user can place a `.obj` file next to the `.stl` to enable Drake, while leaving other users of the URDF unaffected.  I believe users would load the `.stl` into something like `meshlab` and export to `.obj` as a work-around.  It's possibly that ongoing geometry work will improve this situation; I'll assign a couple people who might know more.

christian-rauch, 2018-01-10:
Thanks for the clarification, I'll have use the same workaround then.
Have you thought about using [`assimp`](http://assimp.sourceforge.net) as mesh importer in Drake? It supports many relevant mesh formats.

jwnimmer-tri, 2018-01-10:
I agree with that suggestion.

amcastro-tri, 2018-01-10:
My knowledge on the subject goes pretty much as far as @jwnimmer-tri already expressed above. The workaround @jwnimmer-tri mentions is what we've been doing so far IFAIK. Our geometry guy @SeanCurtis-TRI might be able to give us another quick solution?

Right now we are working on a parser based on the [`sdformat` library](https://bitbucket.org/osrf/sdformat). The `sdformat` library has been integrated into Drake per #4337 and @nkoenig from OSRC is helping us to write a new multibody parser based off `sdformat`. @nkoenig for sure can give us details regarding the supported file formats by `sdformat`.

BTW, that library `assimp` looks good!

SeanCurtis-TRI, 2018-01-10:
I have nothing particular to add to the woes. Short-term, we probably won't see much of a change. Long-term, we'll support more geometric representations.

jwnimmer-tri, 2018-12-21:
Duplicate of #2941.

closed 2018-12-21 by jwnimmer-tri