RobotLocomotion/drake #7734: kuka_plan_runner crashes with binary install, can't find URDF

5yler, 2018-01-11:
Installed drake from `drake-latest-xenial.tar.gz` binary release (accessed Jan 11th 2018) into `/opt/drake`. Trying to run `plan_runner` fails because the URDF is not found:

```
09:37 t2box /opt/drake/share/drake/examples/kuka_iiwa_arm ./kuka_plan_runner 
terminate called after throwing an instance of 'std::runtime_error'
  what():  could not find resource: drake/manipulation/models/iiwa_description/urdf/iiwa14_primitive_collision.urdf
Aborted (core dumped)
```

hongkai-dai, 2019-03-25:
@fbudin69500 could you update the status on this issue? Thanks a lot!

EricCousineau-TRI, 2019-03-25:
Running locally on machine @ 4c6246197 seems to indicate that this is no longer an issue (due to relevant `find_resource` fixes). Closing.

closed 2019-03-25 by EricCousineau-TRI