RobotLocomotion/drake #7747: mac install_prereqs.sh results in obtaining bazel 0.9.0, but we only support 0.6.1

RussTedrake, 2018-01-12:
This might be causing the problem in https://github.com/RobotLocomotion/drake-shambhala/issues/78.  

But really, I think it's more fundamental than a bazel problem.  These installation instructions are not pegging versions, so will move with homebrew.  Are the build servers running install_prereqs regularly?  If yes, then perhaps we primarily need to improve test coverage of the installed packages.

RussTedrake, 2018-01-12:
on the other hand, if CI is moving forward with homebrew (by running install_prereqs often), then it would regularly invalidate documentation like: http://drake.mit.edu/developers.html#supported-configurations

jamiesnape, 2018-01-12:
We regularly update everything but the Bazel version and Xcode. Ultimately, we will be in better shape when Bazel 1.0 is released and there is a little more stability.

jwnimmer-tri, 2018-01-12:
And to be clear, Drake CI _is_ testing bazel 0.9.0 on macOS.  The default builds of master and PRs use 0.6.1, but the Unprovisioned builds run the install_prereqs and then all the tests, e.g., https://drake-jenkins.csail.mit.edu/view/Unprovisioned/job/mac-highsierra-unprovisioned-clang-bazel-nightly-release/75/console.

I don't think we have Shambhala tested on macOS under CI though (especially not as Unprovisioned), which might relate to how the linking oddity was not identified in CI.

jamiesnape, 2018-01-12:
Aside: `brew install bazel@0.5 bazel@0.6  bazel@0.7 bazel@0.8` will get you previous versions from our `robotlocomotion/director` tap (now inappropriately named).

jwnimmer-tri, 2018-11-14:
@EricCousineau-TRI says we should just fix this with a tiny PR that adds an asterisk to the support matrix, saying that the macOS bazel version is what we've tested, but install_prereqs may give you something newer.  Should be quick.

PR #10169 (2019-01-05): doc: Add note about configuration management non-determinism
Resolves #7747

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/10169)
closed 2019-01-05 by EricCousineau-TRI