RobotLocomotion/drake #7751: Create unittests for usage in Jupyter notebooks

EricCousineau-TRI, 2018-01-12:
Specifically, for testing `pydrake` REPL-friendliness,  `call_python_client`, etc.

These may belong in something like `drake-shambhala`.

\cc @rdeits @peteflorence @gizatt

rdeits, 2018-01-12:
BTW, you can use `jupyter nbconvert` with the `--execute` flag to run a notebook from the command line (and exit with an error if any of the cells throws an error).

EricCousineau-TRI, 2018-01-12:
Awesome, thanks!

EricCousineau-TRI, 2018-05-03:
Ended up needing this for some PyTorch stuff, and posted a quick hack here: https://github.com/bazelbuild/rules_python/issues/63

> 
> I quickly put together a simple, kinda hacky, rendition of semi-generalized rules:
> https://github.com/EricCousineau-TRI/repro/tree/52e484e3d68e29bfb8a8e66362c60838161f0dfa/bazel/jupyter_integration
> 
> This allows interactively editing the notebook, using some .runfiles hackery (which I guess may break in upcoming Bazel version) via the ./run script (which perhaps could be replaced via bazel run --direct?).
> You can also run the notebook via bazel run and bazel test.
> 
> Granted, this uses the system jupyter; it focuses on the content of notebooks, not necessarily using a Bazel version of IPython, but I'm sure that could be easy to patch in.

EricCousineau-TRI, 2018-10-11:
Closing in favor of tracking via #9650

closed 2018-10-11 by EricCousineau-TRI