RobotLocomotion/drake #7753: pydrake: Add `RgbdRenderer` for use with `RigidBodyTree` and general shading

EricCousineau-TRI, 2018-01-12:
For simplicity, it'd be nice to move some of the `RigidBodyTree` mechanisms (e.g. `UpdateModelPoses`, visual registration, etc.) to another utility file to minimize the number of components needed to expose to `pybind` for the time being.

\cc @kunimatsu-tri @gizatt and Chris Sweeny (do you have a GitHub acct?)

gizatt, 2018-01-12:
@ChristopherSweeney

EricCousineau-TRI, 2018-01-12:
Question: Would using `RigidBodyTree` work for y'all?
Or would you want to work with raw (Drake) shapes?

C++ `RgbdRenderer` class:
https://github.com/RobotLocomotion/drake/blob/master/systems/sensors/rgbd_renderer.h#L88

(Just using `RigidBodyTree` would simplify what all I need to expose for the time being.)

kunimatsu-tri, 2018-01-12:
My original thought of making `RgbdRenderer` apart from `RgbdCamera` was that being Drake's system and `RigidBodyTree` independent just so that it could be used anywhere as a library. 

So, I'm ok with @EricCousineau-TRI 's proposal as long as the original idea would be preserved.

jwnimmer-tri, 2018-12-21:
@EricCousineau-TRI I'm assuming that we don't care about new RBT features anymore?  Can this be closed?

EricCousineau-TRI, 2018-12-21:
Yup.

FTR, the new abstractions for rendering will be `RenderEngine` for interfacing `SceneGraph` / `GeometryState`.

closed 2018-12-21 by EricCousineau-TRI