RobotLocomotion/drake #7763: //common:drake_assert_test_compile_variants fails with unsupported_crosstool on macOS

jamiesnape, 2018-01-15:
This is an FYI since `unsupported_crosstool` is (inherently) unsupported and we do not ever run the tests using it.

```
FAIL: //common:drake_assert_test_compile_variants (see /private/var/tmp/_bazel_user/7c7ad393d9017d670ea6cd4ae3566d69/execroot/drake/bazel-out/darwin-opt/testlogs/common/drake_assert_test_compile_variants/test.log)
INFO: From Testing //common:drake_assert_test_compile_variants:
==================== Test output for //common:drake_assert_test_compile_variants:
++ find .
.
./tools
./tools/cc_toolchain
./tools/cc_toolchain/osx_cc_wrapper.sh
./common
./common/capture_cc.env
./common/test
./common/test/drake_assert_test_compile_variants.sh
./common/test/drake_assert_test_compile.cc
./common/drake_assert.h
./common/unused.h
./common/drake_assert_test_compile_variants
./third_party
./third_party/com_github_bazelbuild_bazel
./third_party/com_github_bazelbuild_bazel/tools
./third_party/com_github_bazelbuild_bazel/tools/cpp
./third_party/com_github_bazelbuild_bazel/tools/cpp/osx_cc_wrapper.sh
++ capture_cc_env=common/capture_cc.env
++ drake_assert_test_compile_cc=common/test/drake_assert_test_compile.cc
++ source common/capture_cc.env
+++ BAZEL_CC=external/local_config_cc/cc_wrapper.sh
++ [[ ! -z external/local_config_cc/cc_wrapper.sh ]]
++ [[ -x external/local_config_cc/cc_wrapper.sh ]]
++ external/local_config_cc/cc_wrapper.sh --version
/private/var/tmp/_bazel_user/7c7ad393d9017d670ea6cd4ae3566d69/bazel-sandbox/7770256657904808574/execroot/drake/bazel-out/darwin-opt/bin/common/drake_assert_test_compile_variants.runfiles/drake/common/drake_assert_test_compile_variants: line 13: external/local_config_cc/cc_wrapper.sh: No such file or directory
```

FYI @jwnimmer-tri.

EricCousineau-TRI, 2018-05-31:
Per f2f with @jamiesnape, relates #8851: Seems like `$(CC)` may be an untruth of sorts?

jamiesnape, 2018-05-31:
Indeed `export CC=/usr/bin/clang-4.0` => `$(CC) == clang`.

jwnimmer-tri, 2018-11-14:
@jamiesnape Is this test still broken with the latest tooling?

jamiesnape, 2018-11-14:
No.

closed 2018-11-14 by jamiesnape