RobotLocomotion/drake #7775: resource_tool is not linked against libdrake.so

fbudin69500, 2018-01-17:
The best way to find resource files when drake is installed is to find the resources based on their relative directory compared to `libdrake.so`. However, this can only be done if the executable is linked against `libdrake.so`. `resource_tool` is not linked against `libdrake.so` and needs to either use the environment variable (`DRAKE_RESOURCE_ROOT`) or to add search paths to be able to locate resources.

fbudin69500, 2018-01-17:
@jamiesnape @jwnimmer-tri  Just FYI.

jwnimmer-tri, 2018-01-18:
That's a good point.  I am loathe to have `//common`'s BUILD file depend on `libdrake.so` (it seems a bit circular), but probably there is some way to have the installed tool be linked differently (maybe even a separate rule somewhere else that completely recompiles and/or relinks it).

EricCousineau-TRI, 2018-02-20:
Per conversation with @jamiesnape, the PR for this issue should probably remove `drake_RESOURCE_ROOT` from `drake.cps` if it's not longer useful.

EricCousineau-TRI, 2018-12-08:
#8909 may have resolved the linkage issue.

@jamiesnape Should we remove `drake_RESOURCE_ROOT` to close this out?

jwnimmer-tri, 2019-07-02:
I filed #11771 about drake_RESOURCE_ROOT.  And yes, #8909 did resolve the original purpose of this issue.

closed 2019-07-02 by jwnimmer-tri