RobotLocomotion/drake #7779: ImportError: liblcm.so: cannot open shared object file: No such file or directory

airfield20, 2018-01-17:
```
~/Drake$ bazel build automotive:demo automotive:car_sim_lcm
INFO: Analysed 2 targets (1 packages loaded).
INFO: Found 2 targets...
INFO: Elapsed time: 18.599s, Critical Path: 2.79s
INFO: Build completed successfully, 32 total actions
aaron@aaron-VirtualBox:~/Drake$ bazel-bin/automotive/steering_command_driver
Traceback (most recent call last):
  File "/home/aaron/Drake/bazel-bin/automotive/steering_command_driver.runfiles/drake/automotive/steering_command_driver.py", line 18, in <module>
    import lcm
  File "/home/aaron/Drake/bazel-bin/automotive/steering_command_driver.runfiles/lcm/__init__.py", line 1, in <module>
    execfile(__path__[0] + "/lcm-python/lcm/__init__.py")
  File "/home/aaron/Drake/bazel-bin/automotive/steering_command_driver.runfiles/lcm/lcm-python/lcm/__init__.py", line 3, in <module>
    from ._lcm import LCM, LCMSubscription
ImportError: liblcm.so: cannot open shared object file: No such file or directory
```
Manually installing lcm is a workaround

jwnimmer-tri, 2018-01-18:
`bazel run //automotive:steering_command_driver` also fails with the same error as #7712.  Maybe  #7423 broken something.  I'll investigate, fix, and add tests.

jwnimmer-tri, 2018-01-18:
Indeed, `git bisect` blames f50d3b0041a5d5dfee73a3a6ef5035c615dc5592 from #7423.

PR #7788 (2018-01-18): Unbreak python `import lcm` within the sandbox
Fixes #7712.  Fixes #7779.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7788)
closed 2018-01-18 by jwnimmer-tri