RobotLocomotion/drake #7786: Mac install_prereqs.sh upgrades everything

rdeits, 2018-01-18:
I was surprised to see just how long `install_prereqs.sh` takes on mac until I noticed that it runs `brew upgrade`, which upgrades every single homebrew package on the system, including those not used by Drake. This seems like a somewhat unfriendly thing to do without warning, as it could be the source of lots of other subtle versioning issues for packages entirely outside the Drake ecosystem.

rdeits, 2018-01-18:
This also means that *any* failure in the upgrade of any package will cause `install_prereqs.sh` to error out, which has happened several times in a row now for packages completely unrelated to Drake.

jamiesnape, 2018-01-18:
Yes, it is deliberate. Homebrew itself basically relies on having the latest versions of all packages. Invariably Homebrew packages break because you haven't upgraded everything rather than being selective.

PR #8081 (2018-02-15): Refactor install prerequisites scripts
Fixes #7786 among other things. Linux changes are pretty superficial at present. I need to factor out the `apt` packages into a separate file to help with `.deb` packaging, but that will follow later.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8081)
closed 2018-02-15 by jamiesnape