RobotLocomotion/drake #7787: Missing package `@tinyxml` on mac

rdeits, 2018-01-18:
I'm attempting to run the Drake install on OSX 10.11.6, and, after completing the `install_prereqs` step, I'm getting build failures in `@sdformat` relating to a missing `@tinyxml` package:

```
± bazel run //:install -- `pwd`/install 
ERROR: /private/var/tmp/_bazel_rdeits/32256df04d71ea732f8d4ed46baafd3d/external/sdformat/BUILD.bazel:67:1: no such package '@tinyxml//': Unable to complete pkg-config setup for @tinyxml repository: error 1 from [/usr/local/bin/pkg-config, "tinyxml"]:  and referenced by '@sdformat//:sdformat'
ERROR: Analysis of target '//:install' failed; build aborted: no such package '@tinyxml//': Unable to complete pkg-config setup for @tinyxml repository: error 1 from [/usr/local/bin/pkg-config, "tinyxml"]:
INFO: Elapsed time: 1.383s
FAILED: Build did NOT complete successfully (0 packages loaded)
    currently loading: @dreal//
ERROR: Build failed. Not running target
```

`tinyxml` and `tinyxml2` are both installed through Homebrew. Any suggestions for what I should try? Is this potentially related to #6746 ?

soonho-tri, 2018-01-18:
What do you get when you run `pkg-config tinyxml --cflags --libs`?

jwnimmer-tri, 2018-01-18:
Right.  The important part there is `error 1 from [/usr/local/bin/pkg-config, "tinyxml"]` (which makes #6746 unlikely to be related.  That's saying that simply running `/usr/local/bin/pkg-config tinyxml` is returning an error code and no output.  That would indicate that the `*.pc` file is missing (or not on the search path), or that `pkg-config` is broken in some way.

soonho-tri, 2018-01-18:
I think this is not really a macOS 10.11 specific issue. But we know that eventually, you will hit a compilation error (see https://drakedevelopers.slack.com/archives/C0KDGF4V9/p1515772119000100) which is specific to macOS 10.11's Apple-clang compiler. This one itself is not really difficult to patch but please note that we do not support macOS 10.11 anymore. Please consider upgrading to 10.12 or 10.13 when you have time.

rdeits, 2018-01-18:
Indeed, it looks like pkgconfig can't find `tinyxml` (although there is a `/usr/local/lib/pkgconfig/tinyxml.pc` file). Thanks for the update about 10.11. I'll probably just not worry about this for now and leave Drake for my Linux machine.

closed 2018-01-18 by rdeits

soonho-tri, 2018-01-18:
It'd be great if `brew doctor` can catch a problem and help debug the problem.

jwnimmer-tri, 2018-01-18:
The question would be if `PKG_CONFIG_PATH` has `/usr/local/lib/pkgconfig` on it, or whether the `pkg-config` binary has that path already hard-coded in (the man page is confusing on this point).