RobotLocomotion/drake #7791: Some built filenames are too long for ecryptfs users

ggould-tri, 2018-01-18:
ecryptfs (used for homedir encryption on linux systems, required for all TRI computers) limits filenames to 143 characters.

Drake's documentation build results in filenames like "classdrake_1_1examples_1_1kuka__iiwa__arm_1_1box__rotation_1_1_iiwa_and_box_plant_with_state_estimator_a8a9de6aad54e3098b0144c7dd2040317_cgraph.md5" (148 characters).

Not to say that "143 characters ought to be enough for anyone" but...  143 characters ought to be enough for anyone.

jwnimmer-tri, 2018-01-18:
https://www.stack.nl/~dimitri/doxygen/manual/config.html#cfg_short_names

jwnimmer-tri, 2018-01-18:
The question would be if we turn it on always and then our website may have less readable URL slugs, or whether our doxygen.py wrapper should query whether long files are trouble and then enable the option only if so.

jwnimmer-tri, 2018-02-03:
Possibly worth solving in concert with #7920.

jwnimmer-tri, 2018-02-03:
@jamiesnape I'll assign this to you in case the shortname setting helps with #7920; if you end up not touching this issue, you can assign it back to me.

jwnimmer-tri, 2019-10-10:
TRI no longer uses ecryptfs.  I think it would be OK to close this issue?

ggould-tri, 2019-10-10:
TRI still supports ecryptfs homedir encryption -- in fact this is still the case on my desktop machine.

jwnimmer-tri, 2020-04-28:
@ggould-tri Can you check locally wheather generating the built documentation into `/tmp/something` instead of the source tree is a suitable work-around?

jwnimmer-tri, 2021-11-10:
Closing out as "backlog unlikely to reach an actionable state within a reasonable timeframe".  In any case, it's likely that #14834 and alike have resolved this in any case (they build into and serve from a tmpfs).

closed 2021-11-10 by jwnimmer-tri