RobotLocomotion/drake #7796: lcm_vector_gen.py fails because .clang-format already exists

KyleFromKitware, 2018-01-18:
```
09:30:31 ERROR: /home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-coverage/src/automotive/BUILD.bazel:37:1: Couldn't build file automotive/gen/bicycle_car_parameters.cc: SkylarkAction automotive/gen/bicycle_car_parameters.cc failed (Exit 1)
09:30:31 Traceback (most recent call last):
09:30:31   File "/home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-coverage/_bazel_ubuntu/d3d6be48f350ff9f6e7359acf59b6b1d/execroot/drake/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/../drake/tools/vector_gen/lcm_vector_gen.py", line 707, in <module>
09:30:31     main()
09:30:31   File "/home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-coverage/_bazel_ubuntu/d3d6be48f350ff9f6e7359acf59b6b1d/execroot/drake/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/../drake/tools/vector_gen/lcm_vector_gen.py", line 703, in main
09:30:31     return generate_all_code(args.srcs, args.outs)
09:30:31   File "/home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-coverage/_bazel_ubuntu/d3d6be48f350ff9f6e7359acf59b6b1d/execroot/drake/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/../drake/tools/vector_gen/lcm_vector_gen.py", line 686, in generate_all_code
09:30:31     generate_code(src, **kind_to_out)
09:30:31   File "/home/ubuntu/workspace/linux-xenial-clang-bazel-experimental-coverage/_bazel_ubuntu/d3d6be48f350ff9f6e7359acf59b6b1d/execroot/drake/bazel-out/host/bin/external/drake/tools/vector_gen/lcm_vector_gen.runfiles/drake/../drake/tools/vector_gen/lcm_vector_gen.py", line 641, in generate_code
09:30:31     assert not os.path.exists(temp_dotfile)
09:30:31 AssertionError
```

My initial thought is that this could be caused if Bazel tries to build in parallel. If it tries to simultaneously generate and clang-format two files in the same directory, one of the jobs will fail because the other one already created .clang-format in that directory.

KyleFromKitware, 2018-01-18:
This appears to be related to #7340.

jwnimmer-tri, 2018-01-18:
Yeah, this is possibly caused by us not declaring to Bazel that we are going to be creating (and removing) that intermediate file.  I can look into it.

Since builds in CI and locally are passing, I don't know how to reproduce this yet.  Could you post more details about how I could cause this error myself, so that I can explore from there?

KyleFromKitware, 2018-01-18:
I was not able to reproduce this on my local machine either. We're seeing this error fairly frequently (though not always) in the linux-xenial-clang-bazel-experimental-coverage builds in the CI environment. Since we're adding coverage, it may have changed the build timing in such a way that those files get generated in parallel and trigger the error. My advice for reproducing would be to get Bazel to run as many parallel jobs as possible, and add a long wait between `os.symlink()` and `os.unlink()` to make the collision more likely.

Also, what about simply dropping .clang-format into the generated root once instead of linking it for every single file?

jwnimmer-tri, 2018-01-18:
Perhaps the `--spawn_strategy=standalone` for kcov is also a required factor, then.

jwnimmer-tri, 2018-01-18:
I wonder if `--output_user_root` is also in play, if two different Jenkins jobs are stomping on each other, maybe only when `--spawn_strategy=standalone` is used.

jwnimmer-tri, 2018-01-18:
Though I guess only one Bazel job will be running on a machine at a time.  Anyway, I will go now and look at my codegen rules and see if I'm missing some correctness annotations.