RobotLocomotion/drake #7799: Rename @lcm to avoid directory name collisions

jwnimmer-tri, 2018-01-18:
In `WORKSPACE`, we use `@lcm` as the name for the LCM external.  When combined with pythonpath module lookup semantics and native `*.so` loading, this ultimately leads to confusing breaks like #7788.  (See also source code comments nearby there for more details.)

We should explore whether simple using a new name allows us to remove the work-arounds.

EricCousineau-TRI, 2018-01-20:
Potentially related: https://github.com/bazelbuild/bazel/issues/3998
(Had run into this when prototyping `external_data` stuff.)

Should we consider using `--noexperimental_python_import_all_repositories` to avoid this in the future?

jwnimmer-tri, 2018-03-22:
I started a WIP a while ago at https://github.com/jwnimmer-tri/drake/commits/rename-lcm.  I seem to recall it was getting hairy when I paused.

jwnimmer-tri, 2018-05-24:
We can do this if / when we need.  Doesn't seem to be actively causing pain right now.

closed 2018-05-24 by jwnimmer-tri