RobotLocomotion/drake #7826: Determine policies for private linking (static vs. shared, tradeoffs) for upstream libraries

EricCousineau-TRI, 2018-01-22:
(Follow-up from discussion in #7773)

Takeaways:
* If using static libraries with `-fvisibility=hidden`, ensure that all symbols are hidden.
   * May not be the case with templating? (Jamie mentioned that `nm -gU libdrake.so | grep octomap` outputs symbols, which may not be the desired result?)
* If using shared libraries, consider methods to hide the symbols.

\cc @jwnimmer-tri

jwnimmer-tri, 2018-01-22:
Relatedly, I've also had a desire to update `tools/workspace/README.md` for "Adding new third-party software" to help give guidance on how to categorize a new library dependency in terms of its install, and thus what kind of install or link rules we need (or don't need) for them.

jwnimmer-tri, 2021-11-11:
Here's the answer:
- Obey the linting in #15816.
- Host-provided libraries should be dynamic (e.g., libpng).
- Workspace externals build from source should be static, unless the license prevents that (LGPL).

Ideally we would severely narrow the public symbols of `libdrake.so`, but that's a different issue.

closed 2021-11-11 by jwnimmer-tri