RobotLocomotion/drake #7832: pydrake: `undefined symbol: _ZN17btConvexHullShapeC1EPKdii` under Bazel with `--compiler=gcc-5`

EricCousineau-TRI, 2018-01-22:
Relates [this comment](https://github.com/RobotLocomotion/drake/pull/7829#issuecomment-359602234):

On my machine (16.04), I get the following runtime errors:

```shell
$ cd drake
$ git log -n1 --oneline
122e44b Merge pull request #7823 from KyleFromKitware/bazel-coverage
$ bazel test --c dbg --compiler=gcc-5 //bindings/pydrake:common_test
...
ImportError: ${bazel_cache}/execroot/drake/bazel-out/gcc-5-linux-dbg/bin/bindings/pydrake/common_test.runfiles/drake/bindings/pydrake/../../_solib_k8/_U_S_Sbindings_Spydrake_C_Ucommon_Upy.so___Utools_Sinstall_Slibdrake/libdrake.so: undefined symbol: _ZN17btConvexHullShapeC1EPKdii
```

Tried running `install_prereqs.sh`, but did not see any noticeable difference.

jwnimmer-tri, 2018-01-22:
That command passes for me (Bazel 0.6.1 & Ubuntu 16.04).

soonho-tri, 2018-01-23:
Same here. (Bazel 0.9.0 & Ubuntu 16.04)

jamiesnape, 2018-01-30:
I cannot reproduce either and probably would not be able to diagnose with having the binary artifact(s) in question. Tracing though our Bazel logic, I do not see any obvious way this would occur. I am pretty confident, it would not be `install_prereqs.sh` related, but beyond that I do not know.

For future reference, with `libdrake.so` issues, it is helpful to examine the library with `ldd` and `objdump` and post relevant output (in line or as a gist).

I suggest we close this for now and reopen if it reoccurs.

jwnimmer-tri, 2018-04-23:
@EricCousineau-TRI can we close this now?

EricCousineau-TRI, 2018-04-24:
Closing. Missed the comment to close it earlier (sorry about that!) and confirmed that the following test did pass as of `master` (`034ce7e`):
```
bazel test --compiler=gcc-5 //bindings/pydrake:_isolated/forward_diff_test
```

closed 2018-04-24 by EricCousineau-TRI