RobotLocomotion/drake #7844: cannot import pydrake.systems.framework without importing pydrake

RussTedrake, 2018-01-24:
```
Russs-MacBook-Air:src russt$ python
Python 2.7.14 (default, Jan  6 2018, 12:16:16)
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import pydrake.systems.framework
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: arg(): could not convert default argument into a Python object (type not registered yet?). Compile in debug mode for more information.
>>> import pydrake
>>> import pydrake.systems.framework
>>>
```
(it imports without error the second time)

is this dependency expected?  it's new today

RussTedrake, 2018-01-24:
correction -- the import pydrake doesn't help.  it just fails silently the second time.  might it be mac specific?

EricCousineau-TRI, 2018-01-24:
It might be my use of `optional` here:
https://github.com/RobotLocomotion/drake/pull/7838/files#diff-0d8cf8339e9d28cd9e2f70b5359012ddR98

My initial guess is that it's due to `stx::optional` being used rather than `std::optional`, and `pybind` may only be aware of `std::optional`.

Will test it out on Linux + Mac.

EricCousineau-TRI, 2018-01-24:
Marking this closed for now, assuming it is fixed by #7847.

closed 2018-01-24 by EricCousineau-TRI