RobotLocomotion/drake #7855: Humanoid examples fail with Valkyrie mesh path errors

rdeits, 2018-01-25:
Trying to run the humanoid balancing demo currently fails with the following message:

```
± ./bazel-bin/examples/humanoid_controller/valkyrie_balancing_demo
[2018-01-25 11:35:35.207] [console] [warning] Couldn't find package 'valkyrie' in the supplied packagepath.
[2018-01-25 11:35:35.207] [console] [warning] Mesh 'package://valkyrie/urdf/model/meshes/pelvis/pelvis.dae' could not be resolved and will be ignored by Drake. If you don't want it to be ignored, please include it in the package map.
terminate called after throwing an instance of 'std::runtime_error'
  what():  multibody/parsers/urdf_parser.cc: ParseGeometry: ERROR: Mesh file name could not be resolved from the provided uri "package://valkyrie/urdf/model/meshes/pelvis/pelvis.dae".
[1]    1338 abort (core dumped)  ./bazel-bin/examples/humanoid_controller/valkyrie_balancing_demo
```

likewise for the simulation:

```
± bazel-bin/examples/valkyrie/valkyrie_simulation
[2018-01-25 11:34:48.532] [console] [warning] Couldn't find package 'valkyrie' in the supplied packagepath.
[2018-01-25 11:34:48.532] [console] [warning] Mesh 'package://valkyrie/urdf/model/meshes/pelvis/pelvis.dae' could not be resolved and will be ignored by Drake. If you don't want it to be ignored, please include it in the package map.
terminate called after throwing an instance of 'std::runtime_error'
  what():  multibody/parsers/urdf_parser.cc: ParseGeometry: ERROR: Mesh file name could not be resolved from the provided uri "package://valkyrie/urdf/model/meshes/pelvis/pelvis.dae".
[1]    1180 abort (core dumped)  bazel-bin/examples/valkyrie/valkyrie_simulation
```

The mesh file does exist, and it's still at `valkyrie/urdf/model/meshes/pelvis/pelvis.dae`, so it would seem that there's something wrong with the parser's package handling. Any suggestions?

jwnimmer-tri, 2018-01-25:
I can't immediately reproduce this error.  My steps:
- `git checkout master`
- comment-out the `throw std::runtime_error("Gurobi solver not available.");` in `qp_inverse_dynamics.cc`
- `bazel build //examples/humanoid_controller:valkyrie_balancing_demo`
- `bazel-bin/examples/humanoid_controller/valkyrie_balancing_demo`
then I see
```
[2018-01-25 11:54:55.555] [console] [info] Welding joint left_camera_optical_frame_joint
[2018-01-25 11:54:55.555] [console] [info] Welding joint right_camera_optical_frame_joint
```
followed by a long pause.

jwnimmer-tri, 2018-01-25:
My only guess at the moment is that `DRAKE_RESOURCE_ROOT` may be pointing to a problematic copy of the resource, and thus pre-empting use of the copy from the source tree.

I'll come back to this later today and see if there is more debug tracing we can turn on.

rdeits, 2018-01-25:
I haven't run this code since before the `drake` folder hoist. Is there anything special I need to do to account for the hoisting?

jwnimmer-tri, 2018-01-25:
Nope, it should just work (without any environment variables).  Setting `DRAKE_RESOURCE_ROOT` is an override that changes where Drake looks; if you had some old value in your shell environment possibly its confusing things, though.

rdeits, 2018-01-25:
No, I don't have a `DRAKE_RESOURCE_ROOT` environment variable in that shell.

rdeits, 2018-01-25:
This line looks suspiciously like it's searching up the directory tree for a folder called "drake", which no longer exists: https://github.com/RobotLocomotion/drake/blob/7d805aca8d3636bf343ee824362f2ea237cb8f8a/multibody/parsers/package_map.cc#L175

rdeits, 2018-01-25:
Confirmed: changing `kDrakeFolderName` to `drake-distro` fixes the issue. It looks like that constant must be exactly equal to the name under which Drake is cloned.

rdeits, 2018-01-25:
It appears that this is actually a duplicate of https://github.com/RobotLocomotion/drake/issues/7593 . I'll update the title.

jwnimmer-tri, 2018-01-25:
Ah, thanks for diagnosing.

#7593 fixed the `bazel run //examples:foo` form to work; this issue it about fixing the `bazel-bin/examples/foo` form.

jwnimmer-tri, 2018-01-25:
Actually, that wasn't right.  #7593 fixed finding URDFs; this issue is about finding sub-resources like OBJs, based on `package.xml` search heuristics.

jwnimmer-tri, 2018-01-25:
@rdeits I am curious if #7864 fixes it for your case.  It seems to work in my testing.

rdeits, 2018-01-25:
Thanks! I'll try it out as soon as possible.

rdeits, 2018-01-25:
@jwnimmer-tri yes, #7864 seems to fix it for me. Thanks!

PR #7864 (2018-01-26): Fix PopulateUpstreamToDrake for checkout not named drake
The prior version relied on `"/drake/"` appearing in absolute pathnames to URDF files, then looked for `package.xml` files in the trip from the URDF down to drake's root.  This worked when all resources were in `drake-distro/drake`.  When we got rid of `drake-distro`, it still worked if your git clone was named `drake`, or if you were running a test in the bazel sandbox where the workspace is named `drake`.  If you were running from the command line, it would give up because it didn't know where Drake was.

The fix is to use `GetDrakePat [... cut]
closed 2018-01-26 by jwnimmer-tri