RobotLocomotion/drake #7861: `drake_visualizer` cannot find `pydrake` for resolving package paths

rdeits, 2018-01-25:
Observed while running the Valkyrie simulation. `drake_visualizer` (installed by Drake) fails with:

```
INFO: Running command line: bazel-bin/tools/drake_visualizer
Traceback (most recent call last):
  File "/home/rdeits/.cache/bazel/_bazel_rdeits/e8708faf339a91021110cca2d500ef13/execroot/drake/bazel-out/clang-4.0-linux-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/lcmUtils.py", line 119, in handleMessage
    callback(msg)
  File "/home/rdeits/.cache/bazel/_bazel_rdeits/e8708faf339a91021110cca2d500ef13/execroot/drake/bazel-out/clang-4.0-linux-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/drakevisualizer.py", line 344, in onViewerLoadRobot
    self.addLinksFromLCM(msg)
  File "/home/rdeits/.cache/bazel/_bazel_rdeits/e8708faf339a91021110cca2d500ef13/execroot/drake/bazel-out/clang-4.0-linux-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/drakevisualizer.py", line 368, in addLinksFromLCM
    self.addLink(Link(link), link.robot_num, link.name)
  File "/home/rdeits/.cache/bazel/_bazel_rdeits/e8708faf339a91021110cca2d500ef13/execroot/drake/bazel-out/clang-4.0-linux-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/drakevisualizer.py", line 291, in __init__
    self.geometry.extend(Geometry.createGeometry(link.name + ' geometry data', g))
  File "/home/rdeits/.cache/bazel/_bazel_rdeits/e8708faf339a91021110cca2d500ef13/execroot/drake/bazel-out/clang-4.0-linux-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/drakevisualizer.py", line 264, in createGeometry
    polyDataList, visInfo = Geometry.createPolyDataFromFiles(geom)
  File "/home/rdeits/.cache/bazel/_bazel_rdeits/e8708faf339a91021110cca2d500ef13/execroot/drake/bazel-out/clang-4.0-linux-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/drakevisualizer.py", line 196, in createPolyDataFromFiles
    filename = Geometry.resolvePackageFilename(geom.string_data)
  File "/home/rdeits/.cache/bazel/_bazel_rdeits/e8708faf339a91021110cca2d500ef13/execroot/drake/bazel-out/clang-4.0-linux-opt/bin/tools/drake_visualizer.runfiles/drake/external/drake_visualizer/lib/python2.7/dist-packages/director/drakevisualizer.py", line 185, in resolvePackageFilename
    import pydrake
ImportError: No module named pydrake
```

The result is that the visualizer doesn't find any of the Val meshes and thus can't visualize the robot. 

I can reproduce this by building `//...` and running `bazel-bin/tools/drake_visualizer` and `bazel-bin/examples/valkyrie/valkyrie_simulation`. Running `drake_visualizer` with `bazel run` does not appear to make a difference.

EricCousineau-TRI, 2018-01-25:
Hm... Not sure why this is happening.

@jamiesnape Any ideas on this? Is this related to any pre-packaged `drake_visualizer` updates?

jamiesnape, 2018-01-25:
I have not actually updated the packages yet, so no.

EricCousineau-TRI, 2018-01-25:
Alright, I am working on reproducing this locally, will update in this comment once I've finished building off of `master`.

EricCousineau-TRI, 2018-01-25:
@rdeits Just to check, do you still get an error if you use `bazel run //:install`, or if you [use CMake](http://drake.mit.edu/python_bindings.html#building-the-python-bindings)?

rdeits, 2018-01-25:
Thanks @EricCousineau-TRI ! I will try with `//:install` and report back.

rdeits, 2018-01-25:
After `//:install`, the installed `drake-visualizer` does find `pydrake` successfully, but then fails to find the valkyrie meshes (even though I am now running from a checkout named "drake"):

```
failed to resolve filename, unknown package: package://valkyrie/urdf/model/meshes/pelvis/pelvis.dae
warning, cannot find file: package://valkyrie/urdf/model/meshes/pelvis/pelvis.dae
Traceback (most recent call last):
  File "/home/rdeits/locomotion/drake/install/lib/python2.7/dist-packages/director/lcmUtils.py", line 119, in handleMessage
    callback(msg)
  File "/home/rdeits/locomotion/drake/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 344, in onViewerLoadRobot
    self.addLinksFromLCM(msg)
  File "/home/rdeits/locomotion/drake/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 368, in addLinksFromLCM
    self.addLink(Link(link), link.robot_num, link.name)
  File "/home/rdeits/locomotion/drake/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 291, in __init__
    self.geometry.extend(Geometry.createGeometry(link.name + ' geometry data', g))
  File "/home/rdeits/locomotion/drake/install/lib/python2.7/dist-packages/director/drakevisualizer.py", line 264, in createGeometry
    polyDataList, visInfo = Geometry.createPolyDataFromFiles(geom)
ValueError: need more than 0 values to unpack
```

EricCousineau-TRI, 2018-01-26:
Is this related to #7855? Or does this seem like another problem?

rdeits, 2018-01-26:
Unrelated--I've already renamed my checkout to `drake` 😉

EricCousineau-TRI, 2018-01-27:
Sorry for the late follow-up, but it seems like the package path issue should be fixed by https://github.com/RobotLocomotion/drake/pull/7774
My assumption is that this may land within the next few business days.
Do you have a workaround for this, or is it blocking your work?

rdeits, 2018-01-27:
Ok, makes sense, thanks. I can work around it for now.

EricCousineau-TRI, 2018-01-28:
This should be fixed once the two PRs are merged:
- [ ] #7870 
- [ ] #7887