RobotLocomotion/drake #7862: Ipopt fails to solve QP inverse dynamics problem

rdeits, 2018-01-25:
Due to the licensing issues with Gurobi, I'm trying to play with other solvers in the `QPInverseDynamics` code in an attempt to find an acceptable substitute. My first attempt is using Ipopt (since it's widely used in Drake, even if it's not the ideal tool for this), but it's failing immediately:

```
Exception of type: INVALID_TNLP in file "IpTNLPAdapter.cpp" at line 596:
 Exception message: There are inconsistent bounds on constraint 89: lower =   3.4584595208887258e-323 and upper =    0.0000000000000000e+00.
[2018-01-25 16:14:48.193] [console] [warning] Solution not found.
terminate called after throwing an instance of 'std::runtime_error'
  what():  QpInverseDynamicsSystem: QP cannot solve
```

`3.4584595208887258e-323` looks a lot like uninitialized memory being interpreted as a `double`. Looking at the constraints being loaded into Ipopt shows a lot of similar uninitialized-looking data. I'll see if I can figure out what's going on.

rdeits, 2018-01-25:
The issue appears to be that the final constraint in the mathematical program in `qp_inverse_dynamics.cc` has bounds of length 15 but reports its `num_constraints()` as 36. This results in `ipopt` reading out-of-bounds data when assembling the bounds of its NLP. I have added a `DRAKE_DEMAND` that demonstrates the problem in https://github.com/rdeits/drake/tree/rd-no-gurobi

To reproduce, you need to launch `examples/humanoid_controller/valkyrie_balancing_demo` (this will run indefinitely) and then launch `examples/valkyrie/valkyrie_simulation`, which will cause the demo to immediately abort at the `DRAKE_DEMAND` in `systems/controllers/qp_inverse_dynamics/qp_inverse_dynamics.cc:622`. 

My guess is that a linear constraint is being added with a coefficient matrix whose size does not match its bound vectors, but I'm not sure yet. We were probably just getting lucky that this worked in Gurobi because the `GurobiSolver` never calls `num_constraints`.

rdeits, 2018-01-25:
The following test demonstrates the issue:

```
  Eigen::Matrix2d Q2;
  Eigen::Vector2d b2;
  // clang-format off
  Q2 << 1, 0,
        0, 1;
  // clang-format on
  b2 << 1, 2;
  LinearEqualityConstraint constraint(Q2, b2);
  EXPECT_TRUE(constraint.lower_bound().size() == 2);
  EXPECT_TRUE(constraint.upper_bound().size() == 2);
  EXPECT_TRUE(constraint.num_constraints() == 2);

  Eigen::MatrixXd Q3(3, 2);
  Eigen::Vector3d b3;
  // clang-format off
  Q3 << 1, 0,
        0, 1,
        0, 0;
  // clang-format on
  b3 << 1, 2, 3;

  constraint.UpdateCoefficients(Q3, b3);
  EXPECT_TRUE(constraint.lower_bound().size() == 3);
  EXPECT_TRUE(constraint.upper_bound().size() == 3);
  EXPECT_TRUE(constraint.num_constraints() == 3);
```
The underlying cause is that `UpdateCoefficients` does not change the output of `num_constraints()` even if the number of rows is changed.

hongkai-dai, 2018-01-25:
Thanks for looking into this! So `UpdateLowerBound`, `UpdateUpperBound` and `UpdateBounds` also will cause the problem, since they do not update `num_constraints_`.

PR #7892 (2018-01-30): Update num_constraint when updating the bounds in the constraint
Fix #7862 

@rdeits

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7892)
closed 2018-01-30 by hongkai-dai