RobotLocomotion/drake #7871: Consider setting `--noexperimental_python_import_all_repositories` for Drake

EricCousineau-TRI, 2018-01-26:
From https://github.com/robotlocomotion/drake/pull/7848#issuecomment-360823752:

> BTW Should we consider turning on --noexperimental_python_import_all_repositories by default?
bazelbuild/bazel#3998
> I ran into this issue when generating a solution for #7861 (where @vtk got precedence over @vtk//lib/python2.7/site-packages).
> Relates #7799.

PR solution for #7861 is at #7870.

jamiesnape, 2018-01-26:
Flag changed in 0.7. Mac is still on 0.6.1. We should probably wait until the `experimental` aspect is gone anyway unless it critically blocks something.

jwnimmer-tri, 2020-04-28:
@EricCousineau-TRI do you happen to know if this is still relevant with current bazel (3.1)?

jamiesnape, 2020-04-29:
[That flag is listed as undocumented](https://github.com/bazelbuild/bazel/blob/a666f0027adcf2c1cbc5bb314645282289abc6c0/src/main/java/com/google/devtools/build/lib/bazel/rules/python/BazelPythonConfiguration.java#L84) (now), so I figure we should close this anyway.

EricCousineau-TRI, 2020-04-29:
Aye, sounds good.

I think another thing (to avoid clashes) is to possibly investigate is disabling `legacy_create_init`, as mentioned here:
https://github.com/bazelbuild/bazel/issues/7653

However, that will break the workaround for `@drake//bindings/pydrake` (#8240).

I think I had intended to cite this to, but did not way back when. Don't plan on investigating it for now, so will leave issue closed, just with this breadcrumb.

EricCousineau-TRI, 2020-05-14:
@pvarin just got bit by this: https://stackoverflow.com/q/61802860/7829525

Re-opening it, and re-scoping towards somehow removing the init files.

Regarding:

> However, that will break the workaround for @drake//bindings/pydrake (#8240).

This statement is actually false. It can be solved using `Rlocation`.

EricCousineau-TRI, 2020-05-14:
Hm... I guess the shift from "repositories on path" to "no autogen __init__.py files" may have been a bad rescoping. Meh for now.

jwnimmer-tri, 2020-05-14:
Please open a new issue with steps to reproduce, expected result, actual result, etc.  I'm unable to follow the above, even after clicking through to SO.

EricCousineau-TRI, 2020-05-14:
Done: https://github.com/RobotLocomotion/drake/issues/13320

closed 2020-05-14 by EricCousineau-TRI