RobotLocomotion/drake #7875: Running modern Drake on the Atlas robot

rdeits, 2018-01-26:
I'm opening this issue as a way to track the changes that would be necessary to get a modern Drake controller running on the Atlas humanoid. So far this has turned out to be more difficult than I'd hoped, so I think it's worth laying out exactly what would be required. 

- [ ] Visualization
  - [ ] currently does not work due to #7861, but I'm sure this can be fixed.
- [ ] QP Solver
  - [ ] The current QP inverse dynamics uses Gurobi. Unfortunately, Gurobi's academic license does not work inside Docker containers (it's explicitly forbidden). That's a problem for us because the Atlas onboard computers run Ubuntu 14.04, which Drake does not support, and we had planned to use Docker onboard to let Drake run in a 16.04 environment. A few options:
    - [ ] Replace Gurobi with another QP solver. I have tried Ipopt and SCS already (since Drake provides them), but neither could solve the inverse dynamics problems. #7862 is relevant but not a blocker.
    - [ ] Upgrade the Atlas onboard computers to 16.04. We can do this, but I'm quite worried about unforeseen consequences on a system we no longer have official support for. 
    - [ ] Don't use Drake's QP solver, but run the old QP solver from openhumanoids (which runs on 14.04) or some other QP implementation instead.
- [ ] Walking. Walking support is not required, but would obviously be nice. Drake lost its walking planning machinery when the Matlab code was deleted. There's a new C++ implementation in openhumanoids, but it's using an older QP controller. Porting that would be possible but isn't a blocker.

hongkai-dai, 2018-01-26:
There is a github issue #7825 related to the QP solver

rdeits, 2018-01-26:
Ah, yes, thanks!

hongkai-dai, 2018-01-29:
@rdeits , you mentioned that SCS does not work for your QP problem. Does the solver fail to find a solution (reach iteration limits, numerical errors), or it finds a solution, but the accuracy of the solution is too low?

rdeits, 2018-01-29:
SCS actually errors out saying that `Y is not positive definite`, but I haven't investigated further.

RussTedrake, 2019-09-19:
Sadly, this is probably a "won't fix"

closed 2019-09-19 by RussTedrake