RobotLocomotion/drake #7882: Support for Ubuntu Bionic 18.04

jwnimmer-tri, 2018-01-27:
I wanted to start to get this on the radar.  At some point, we are going to need to add official Drake support for Ubuntu Bionic 18.04, at least in CI.  (I don't suspect we'll need binary releases initially.)

Its probably worth seeing what Jenkins / AMI / etc. support we'll need to put in place in order to have at least a couple Bionic builds available in non-production continuous and opt-in experimental when April rolls around.  My guess would be by around May or June we'd prefer to have at least Continuous Production enforcing Bionic support.

Certainly as TRI starts to use Bionic we can test locally and PR fixes before any CI is here, but in case the Jenkins side needs some planning, I thought I'd open a log here to discuss.

jamiesnape, 2018-01-29:
We should be able to get an AMI on release day. We would need builds of Drake Visualizer, VTK, anything in the `drake-apt` repo.

jwnimmer-tri, 2018-01-29:
FYI If it made things easier to have the first milestone with only cut-down version of Drake that excluded those dependencies (e.g., just `test --config no_dreal //systems/anaysis/... //solvers/...`) that would be the most immediately useful part (in ~May), because some TRI projects (with a different WORKSPACE) don't use the pre-builts anyway.  But possibly just anyway making the pre-builts is easier than fiddling with the the configs to omit them.

jamiesnape, 2018-01-29:
It is probably a good carrot to getting all the binaries building automatically on Jenkins, which has forever been on my TODO list.

Much depends on what versions of the various system dependencies they are bundling (I have not checked), but give we use recent versions on Mac, we at least have something vaguely canary-like. If everything compiles without too much trouble, it should be no big deal. Drake Visualizer is probably the only one I would be worried about.

jamiesnape, 2018-04-30:
Logistics-wise, how do we want to do this? Do we mind have mentions of "bionic" in this repo before it is supported or would working on a branch/fork work better (being able to have an `install_prereqs` for 18.04 would make configuring CI somewhat easier, for instance).

jamiesnape, 2018-05-10:
@soonho-tri Could you create some dReal and IBEX debs for Bionic when you have a spare moment? Not urgent.

soonho-tri, 2018-05-16:
@jamiesnape , will send you one this week.

dmsj, 2018-06-13:
What is the status on 18.04 support? I ran into a few issues on quick cursory build attempt.
```bash
~/drake$ bazel build //...
ERROR: /home/dexai/drake/common/BUILD.bazel:402:1: no such package '@gflags//': could not determine Linux distribution: unsupported distribution 'Ubuntu 18.04' and referenced by '//common:text_logging_gflags'
ERROR: Analysis of target '//common/test_utilities:drake_cc_googletest_main' failed; build aborted: no such package '@gflags//': could not determine Linux distribution: unsupported distribution 'Ubuntu 18.04'
INFO: Elapsed time: 2.468s
```
The obvious super simple fixes didn't work.

jamiesnape, 2018-06-13:
Still unsupported, no ETA for support at present.

jwnimmer-tri, 2018-06-13:
Maybe I should PR the `os.bzl` change that enables 18.04 identification, but still leaves any workspace entries that are sensitive to the 16.04 vs 18.04 distinction as failures.  That would at least enable off-label bazel builds on 18.04 for the subset of Drake that doesn't depend on the Drake-specific precompiled dependencies.

jamiesnape, 2018-06-13:
Sure, I have no problem with that.

dmsj, 2018-06-13:
I think we can make it work with the updated `os.bzl` - much appreciated!

jwnimmer-tri, 2018-09-06:
@EricCousineau-TRI I would like to see some traction on this in the next month or two.  There will sure be a Bionic-only userbase within TRI by end of year.

jamiesnape, 2018-09-11:
We will certainly have support on CI by the end of the year, but we are budget limited in terms of both compute and manpower, so it will take time. If you want to increase the priority on this, please do (it is marked as low), but there are challenges regardless.

EricCousineau-TRI, 2018-12-03:
This appears to be completed as Bionic is on CI and is on the support matrix. (Bionic may actually be more covered than Xenial.)

Other issues are documented separately.

closed 2018-12-03 by EricCousineau-TRI