RobotLocomotion/drake #7893: rbtree_py: Ensure this plays nicely with existing bindings

EricCousineau-TRI, 2018-01-29:
Unfortunately, the C++ standards committee did not want to ensure that a `shared_ptr<T>` could be converted to `unique_ptr<T>`.

Since the existing `RigidBodyTree` / `rbtree_py` API uses `shared_ptr<T>`, it makes it a little hard to mix with call-sites that only accept `unique_ptr` (e.g. `RigidBodyPlant`).

The easy solution is to change the constructor of `RigidBodyPlant` to use `shared_ptr<T>` rather than `unique_ptr<T>`, since you can implicitly convert from `unique_ptr<T>` to `shared_ptr<T>`.

jwnimmer-tri, 2018-01-29:
Could you elaborate more about what the problem is?  I see `findFrame` uses `shared_ptr` but I don't see how that relates to the plant or construction-time challenges, and I don't see other uses of shared_ptr that seem problematic?

(Perhaps the python binding for findFrame should just de-ref the shared_ptr and return a normal pointer, anyway?)

EricCousineau-TRI, 2018-01-29:
Hup, you're right. I had been under the impression that `RigidBodyTree` uses `shared_ptr` as its `pybind11` holder type, but ends up that's only `RigidBodyFrame` (and the solvers framework):
https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/rbtree_py.cc#L31
(`py::class_<T>` -> `py::class_<T, unique_ptr<T>>`, where the parameter that is not a base of / nor derived from T is generally interpreted as the holder).
https://github.com/RobotLocomotion/drake/blob/master/bindings/pydrake/rbtree_py.cc#L201

closed 2018-01-29 by EricCousineau-TRI