RobotLocomotion/drake #7895: Drake interferes with Eigen dependencies in Catkin

calderpg-tri, 2018-01-29:
If `<drake_dir>/lib/cmake` is in the `CMAKE_PREFIX_PATH`, Catkin packages declaring Eigen as a dependency will fail to run CMake with the following errors:

```
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_libraries.cmake:137 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION" is not allowed.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:258 (catkin_replace_imported_library_targets)
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  drake_eigen_test/CMakeLists.txt:8 (catkin_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_libraries.cmake:141 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_CONFIGURATIONS" is not allowed.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:258 (catkin_replace_imported_library_targets)
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  drake_eigen_test/CMakeLists.txt:8 (catkin_package)

CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_libraries.cmake:143 (get_target_property):
  INTERFACE_LIBRARY targets may only have whitelisted properties.  The
  property "IMPORTED_LOCATION_Eigen3::Eigen_imported_configurations-NOTFOUND"
  is not allowed.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:258 (catkin_replace_imported_library_targets)
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  drake_eigen_test/CMakeLists.txt:8 (catkin_package)
```

This issue is present on Ubuntu 16.04+ROS Kinetic (and likely other platforms with Catkin). To reproduce:

1. Source ROS: `~$ source /opt/ros/kinetic/setup.bash`
2. Make drake part of the `CMAKE_PREFIX_PATH`: `~$ export CMAKE_PREFIX_PATH=<drake_dir>/lib/cmake:$CMAKE_PREFIX_PATH`
3. Build any Catkin package that declared a dependency on Eigen, such as the example `package.xml` and `CMakeLists.txt` below. 

Example `package.xml`:

```
<?xml version="1.0"?>
<package format="2">
  <name>drake_eigen_test</name>
  <version>0.0.0</version>
  <description>The drake_eigen_test package</description>

  <maintainer email="todo@todo.todo">todo</maintainer>

  <license>TODO</license>

  <buildtool_depend>catkin</buildtool_depend>

  <export>
  </export>
</package>
```

Example `CMakeLists.txt`:

```
cmake_minimum_required(VERSION 2.8.3)
project(drake_eigen_test)

find_package(catkin REQUIRED)
find_package(Eigen3 REQUIRED)
set(Eigen3_INCLUDE_DIRS ${EIGEN3_INCLUDE_DIR})

catkin_package(DEPENDS Eigen3)
```

jamiesnape, 2018-01-29:
Technically a bug in Catkin, but we will have to deal with it.

stonier, 2018-01-30:
@j-rivero can you take a look at this and determine if anything upstream in catkin needs to be remedied?

We should also move on the drake_ros layer and get that tested, used and documented somewhere.

jamiesnape, 2018-01-30:
To elaborate, looks like Catkin is trying to write or re-write properties of an `INTERFACE` library. I have not looked deeply enough as to why it does that, but it is not in general a valid thing to do since by their nature `INTERFACE` libraries only support a small number of properties.

j-rivero, 2018-01-30:
> @j-rivero @nuclearsandwich can you take a look at this and determine if anything upstream in catkin needs to be remedied?

Yep. We think that Jamie is right about being a catkin problem and it could be the bug reported in https://github.com/ros/catkin/pull/916 . I've verified that patching catkin with the patch proposed in the PR solves the compilation problem.

jamiesnape, 2018-02-02:
Fixed in [catkin 0.7.11](https://github.com/ros/catkin/compare/0.7.10...0.7.11)?

j-rivero, 2018-02-07:
> Fixed in catkin 0.7.11?

Right. The problem is that a new catkin releases will hit kinetic infrequently due to the necessary effort rebuild "everything" and for every user to download all packages on an update, so it is still not released for Kinetic yet. The estimated time would be around two weeks from now. If you need it before, I can build a package for you and you place in your CI or repos as a temporary measure.

calderpg-tri, 2018-02-07:
I think having a deb package of the fixed Catkin for our work would be ideal, so long as it isn't going to break something when used with an otherwise up-to-date ROS binary installation.

j-rivero, 2018-02-07:
> I think having a deb package of the fixed Catkin for our work would be ideal, so long as it isn't going to break something when used with an otherwise up-to-date ROS binary installation.

Ready: https://build.osrfoundation.org/view/All/job/catkin-bloom-debbuilder/2/
I've tested it locally and it works fine with the initial example that demonstrated the bug.

I've checked with the catkin maintainer that (unless hidden bugs) it should be perfectly compatible with the rest of the ROS ecosystem.

RussTedrake, 2018-09-23:
@calderpg-tri -- is this now safe to close?

EricCousineau-TRI, 2018-12-08:
Friendly ping?