RobotLocomotion/drake #7903: Add a version (or timestamp or commit SHA) file to the install

jamiesnape, 2018-01-30:
Per discussion with @RussTedrake and @stonier, we need a file that is embedded in the install, so that we can identify the version of an arbitrary installed version of Drake. At its simplest, this could be a file `share/doc/drake/VERSION` with a nightly timestamp or commit SHA. More sophisticated would be a nice header than could also be queried programmatically. Once we have `major.minor.patch.tweak` versioning (which we could be now with `0.x.x.yyyymmdd` if we really wanted), CMake would also want to know this information.

jamiesnape, 2018-01-30:
(Note if we do this within this repo using Bazel we have to be careful not to break caching of the rest of the build. Not difficult, but something to remember.)

jamiesnape, 2018-01-30:
@EricCousineau-TRI To answer the question in #7906, if you have time today, please take this issue, otherwise I will work on it tomorrow morning.

jwnimmer-tri, 2018-01-30:
Maybe somewhere around https://github.com/RobotLocomotion/drake-ci/blob/master/driver/configurations/bazel/step-build.cmake#L108 we could output the date (and also `git rev-parse HEAD`) to an installed file such as `share/drake/version.txt`?

jamiesnape, 2018-01-30:
That is the quick version, yes.

jamiesnape, 2018-01-30:
RobotLocomotion/drake-ci@3d3bb24f9a6f6162c7e48d5fdc9bacea89eab524. I will leave this open at least until I have checked the nightly builds tomorrow morning.

jamiesnape, 2018-01-31:
Nightlies are all good and contain `/opt/drake/share/doc/drake/VERSION.txt` with `<date> <commit>` as the contents.

closed 2018-01-31 by jamiesnape

EricCousineau-TRI, 2018-01-31:
Quick follow-up question: Is it easy to programmatically query this information? Is it easily discoverable via `FindResource`?

e.g. in Python, it'd be nice to have something like `dump_debug_info()`, which would print version, platform, etc., when students have issues.

jamiesnape, 2018-01-31:
> Quick follow-up question: Is it easy to programmatically query this information? Is it easily discoverable via FindResource?

Not at all. It is human readable information for now.

jamiesnape, 2018-01-31:
(Create a follow up issue if necessary.)

EricCousineau-TRI, 2018-01-31:
Yup, shall do.