RobotLocomotion/drake #7904: Have a separate `Simply`

soonho-tri, 2018-01-30:


soonho-tri, 2018-01-30:
/cc @jwnimmer-tri

soonho-tri, 2018-01-30:
I think the first thing we need to do is to remove unsound simplification rules from `symbolic::Expression` implementation. There exist `TODO`s already in the code.

Note that the simplification rule that caused the problem in @jwnimmer-tri 's example is an unsound one:

```
    E * 0.0   -->  0.0
```

Consider the following example to see why this one is unsound. When `E = x / y`, the unsimplified term `(x / y * 0.0)` is evaluated to `nan` when `x = 0.0` and `y = 0.0` while it would be simplified to `0.0`.

Essentially any simplification rule that is eliminating a generic symbolic term `E` is considered unsound. Note that constant folding (i.e. `3 + 4 => 7`) is still considered as a sound one.

I think only eliminating the unsound simplification rules but keeping the sound ones can solve the problem that we encountered in the symbolic sparsity analysis.

jwnimmer-tri, 2018-01-30:
That sounds plausible.  I guess I would have to see what simplifications remained to know for sure.  Certainly if they are only doing constant-folding when the Expression has no Variables, its likely fine.

jwnimmer-tri, 2026-09-10:
At the moment, we don't have plans to further improve the symbolic engine. In general, I suggest using the sympy compatibility functions (pydrake.symbolic.to_sympy, etc.) instead for a more full-featured symbolic manipulation engine.

closed 2026-09-10 by jwnimmer-tri