RobotLocomotion/drake #7908: Fix rgbd_renderer_vtk_test failure

kunimatsu-tri, 2018-01-30:
`bazel test systems/sensors:rgbd_renderer_vtk_test` fails on some machines.

This is probably related to off screen rendering with VTK.
In [VTK's example page](https://cmake.org/Wiki/VTK/Examples/Cxx/Utilities/OffScreenRendering), there is a description saying "You must turn on VTK_OPENGL_HAS_OSMESA in the VTK advanced build configuration", but our VTK doesn't have it.

The current workaround is setting [`kShowWindow`](https://github.com/RobotLocomotion/drake/blob/master/systems/sensors/test/rgbd_renderer_test_util.h#L107) to `true` if you want to run the unit test locally.

rpoyner-tri, 2018-02-02:
in master as of 
```
commit 0ea31a587055139aa4e3433d7a3dd1669a208e8e
Merge: 761780a effe6c1
Author: Eric Cousineau <eric.cousineau@tri.global>
Date:   Thu Feb 1 20:29:29 2018 -0500
```
I had to fix show-window stuff in three places to get `bazel test //...` to pass:

```
diff --git a/systems/sensors/BUILD.bazel b/systems/sensors/BUILD.bazel
index 79fdefa..8d1a4be 100644
--- a/systems/sensors/BUILD.bazel
+++ b/systems/sensors/BUILD.bazel
@@ -482,7 +482,7 @@
         # "test/models/sphere.sdf" to the path below.
         "--sdf_dir=" + native.package_name() + "/test/models",
         "--duration=0.1",
-        "--show_window=false",
+        "--show_window=true",
     ],
     data = [
         ":test_models",
diff --git a/systems/sensors/test/rgbd_camera_test.cc b/systems/sensors/test/rgbd_camera_test.cc
index 99e31ef..07a3a62 100644
--- a/systems/sensors/test/rgbd_camera_test.cc
+++ b/systems/sensors/test/rgbd_camera_test.cc
@@ -42,7 +42,7 @@ namespace {
 // Linux and Mac OSX.
 constexpr double kTolerance = 1e-12;
 constexpr double kFovY = M_PI_4;
-constexpr bool kShowWindow = false;
+constexpr bool kShowWindow = true;
 constexpr double kDepthRangeNear = 0.5;
 constexpr double kDepthRangeFar = 5.;
 constexpr int kWidth = 640;
diff --git a/systems/sensors/test/rgbd_renderer_test_util.h b/systems/sensors/test/rgbd_renderer_test_util.h
index 719139e..590baab 100644
--- a/systems/sensors/test/rgbd_renderer_test_util.h
+++ b/systems/sensors/test/rgbd_renderer_test_util.h
@@ -104,7 +104,7 @@ class RgbdRendererTest : public ::testing::Test {
   const double kZNear = 0.5;
   const double kZFar = 5.;
   const double kFovY = M_PI_4;
-  const bool kShowWindow = false;
+  const bool kShowWindow = true;
 
   // The following tolerance is used due to a precision difference between
   // Ubuntu Linux and Mac OSX.
```

Can we get the number of places to fix down to one, and/or come up with some platform sniffing in the build system that can set the flag automagically?

jamiesnape, 2018-02-02:
> Can we get the number of places to fix down to one, and/or come up with some platform sniffing in the build system that can set the flag automagically?

Probably. Can you create a separate issue?

rpoyner-tri, 2018-02-02:
new issue done.

EricCousineau-TRI, 2018-02-08:
Is this at all related to [this FAQ entry](http://drake.mit.edu/faq.html#why-do-opengl-based-vtk-targets-run-with-bazel-test-sometimes-fail-on-linux), created by #7004?

jamiesnape, 2018-02-08:
`rgbd_renderer_vtk_test` is already marked `local`, so I imagine not.

PR #8789 (2018-05-16): Update to VTK 8.1 and latest Director and add packages for Bionic
Needs local testing on Linux of `drake-visualizer` using both `bazel run //tools:drake_visualizer` and the installed executable (`bazel run install -- /opt/drake && /opt/drake/bin/drake-visualizer`). Note that for licensing reasons this switches to using a shared library for LCM (c.f., #7423, #8153).

Fixes #7908, fixes #8322, fixes #8764. Relates #7907.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/8789)
closed 2018-05-16 by sammy-tri