RobotLocomotion/drake #7914: pydrake: Provide documentation.

EricCousineau-TRI, 2018-01-31:
Presently, most of the bindings of `pydrake` does not provide documentation.

My ideal solution would be to have some sort of link between the Python code and the C++ code, so that we don't waste time duplicating documentation. I assume this could be automated, and applied automatically (either at compile-time, or at run-time), such that `help(...)` in Python / IPython is informative to users.

The alternative is to manually duplicate all documentation.
This is much less preferable for me, and seems less maintainable.

\cc @naveenoid

jwnimmer-tri, 2018-01-31:
FWIW I agree that mostly the python documentation should (semi-?)automatically come over from the C++ documentation.  However, we should still encourage people to add (or supplement) that with python-specific help strings where it makes sense.  (And pure-python non-private code should always have docstrings, per the style guide -- a few small places in bindings are currently missing it.)

jwnimmer-tri, 2018-03-17:
The `GENERATE_XML = Yes` output of Doxygen seems plausibly parse-able at runtime via Python.  The first approach I'd try exploring is to have the `help()` query lazily load the Doxygen XML output and find the relevant section(s) to show, with only a loose dependency so Doxygen isn't in the critical path for locally running and testing the bindings.

EricCousineau-TRI, 2018-03-17:
Ah, didn't think about lazy evaluation!
From a quick search, though, and from what I've dealt with with descriptors and `__dict__` entries (which `__doc__` is), it seems that this would require additional effort:
https://stackoverflow.com/a/2694358/7829525
We could monkey patch the `pybind11` metaclass, but then we still have functions, methods, and modules to patch. We could wrap those with `wrapt` or what not to proxy (everything, but namely) `__doc__`, which should work -- but I smell a bit of storm brewing in that solution. Will still keep my eyes peeled, though!

Overall, I think that this approach would require a lot more fine-tuning in implementation, whereas the non-lazy scanning approach could be a first pass that, if implemented well, could then pave the way for dynamic docstrings if it's apparent they're necessary for performance / usability.

> [...] with only a loose dependency so Doxygen isn't in the critical path for locally running and testing the bindings.

For the non-lazy patching approach, my intent is to definitely do a loose dependency at runtime: have some method, like `pydrake::UseDoxygenIfAvailable(m)`, which, if the doxygen XML was available, would scan the entries in the current module, find the relevant entry, and patch them.

Side note:
When perusing the `pybind11` project, seems like there's a codegen-related option of sorts already out there (if the other options are not viable):
https://github.com/pybind/pybind11/pull/1321

EricCousineau-TRI, 2018-03-17:
Also, I'd probably end up using `breathe` or importing a portion of it - at least for a first pass - since they've already done the work of parsing it for mapping to Sphinx / rst:
https://github.com/michaeljones/breathe

EricCousineau-TRI, 2018-07-31:
`pybind` does have a utility for creating C++ `#define` statements with the docstrings:
https://github.com/pybind/pybind11/blob/master/tools/mkdoc.py

Haven't looked much deeper into it, though.

fbudin69500, 2018-08-16:
@EricCousineau-TRI As a follow up of our conversation this morning, ITK does basically what @jwnimmer-tri suggested to add its doxygen documentation to its Python bindings:
* https://github.com/InsightSoftwareConsortium/ITK/blob/master/Wrapping/Generators/Doc/CMakeLists.txt
* The generated XML files are converted to *.i files that SWIG, which is used by ITK to create its Python wrapping, understands.
* In the past, the second step was not performed, and the documentation was loaded at runtime if it was available (if the files were found in a specific directory). One could indeed use this solution, at least at first, and install the documentation files during drake installation process.

EricCousineau-TRI, 2018-08-29:
Per Slack chat with @m-chaturvedi, he's trying out the `mkdoc.py` approach, and will see what this approach yields.

For testing, we made a simple trimmed-down branch:
https://github.com/EricCousineau-TRI/drake/tree/feature/py_just_common

m-chaturvedi, 2018-09-05:
The "`mkdoc.py` approach" is outlined here by a contributor of Pybind11:
https://github.com/pybind/pybind11/issues/807#issuecomment-295024217

`mkdoc.py` is calling clang internally to parse the header files so it needs the include directories just like we need while building.

The two main problems it presents us are the following:
1.  Given a `.cc` file, example `math_py.cc`, get the header files from it (non-transitive, example `#include "drake/math/barycentric.h"`) on which `mkdoc.py` needs to be run.  
2. On the header files run `mkdoc.py`.  Running `mkdoc.py` needs the `-I` similar to how `gcc` needs `-I`s while building.

jamiesnape, 2018-09-05:
> On the header files run mkdoc.py. Running mkdoc.py needs the -I similar to how gcc needs -Is while building.

More precisely, it is running `clang` under the hood so it needs include directories.

jwnimmer-tri, 2018-09-05:
Since pydrake is using `libdrake.so` for basically all of the C++ code that's going to be wrapped, I'm not sure why we'd need fine-grained parsing of `math_py.cc`'s includes like `barycentric.h`.  There could be a single build rule that processed all of the `//:drake_shared_library` headers into a single docstring `*.inc` file, which any `math_py.cc` that wanted to access the comments could use.

jamiesnape, 2018-09-05:
We certainly discussed having one doc header per module, but that ends up being a small implementation detail. Neither of what @m-chaturvedi mentions are really "problems" in the sense that they are difficult.

jamiesnape, 2018-09-17:
When processing all the `//:drake_shared_library` headers, clang is choking on the various includes of `drake/common/autodiff.h` and `drake/common/symbolic.h`.

EricCousineau-TRI, 2018-09-18:
Can I ask in what way it's choking? Is there visible output on it, or does it just hang forever as it expands a massive amount of template instantiations?

jamiesnape, 2018-09-18:
Numerous compilation errors. Not sure if it is because no-one compiles those headers in isolation now, or some other reason.

jamiesnape, 2018-09-18:
I have a feeling the choice of `#pragma once` is part of the issue since I am seeing `-Wpragma-once-outside-header` warnings and multiple definitions of various types. I will see if adding some (redundant) include guards helps.

jamiesnape, 2018-09-18:
> I have a feeling the choice of #pragma once is part of the issue [...]

Yes, that one of the problems. Since I presume #2104 is irreversible, I will add redundant guards (at least in the short term).

Meanwhile, some of the  `drake/common/autodiff.h` are not compilable without adding some includes in the `#ifndef  DRAKE_COMMON_AUTODIFF_HEADER ... #endif` that contains the `#warning`.

jamiesnape, 2018-09-18:
Finding a lot of headers that do not IWYU (or even include-what-you-need). Some clearly deliberate, but not all. I guess this is a signal that I should be dusting off my branch for running IWYU (easiest when we switch to Bionic).

jwnimmer-tri, 2018-09-18:
What is this header reprocessing tool doing, that diverges from what the compiler (with preprocesser + etc) would be doing while parsing a header?  The compilers seem to like the headers; why is this tool different?  Of course if there are code defects we should fix them anyway, but I worry a bit about adopting a tool which diverges from what compilers do, in terms of ongoing upkeep.

jamiesnape, 2018-09-18:
It is a compiler, but it only processes headers.

jamiesnape, 2018-09-18:
(When headers are treated like source files, `#pragma once` does not work. Otherwise, everything else is what I would consider a defect.)

jwnimmer-tri, 2018-09-18:
... but doesn't process include statements?  Otherwise, how do IWYU violations cause it to fail?  (Presumably, some transitively included header got the job done, for regular compilers?)

jamiesnape, 2018-09-18:
It processes `#include` statements, but we have some headers that apparently are never normally used in isolation.

jwnimmer-tri, 2018-09-18:
Ah, okay.  Most headers have a matching cc file or unit test that includes them standalone as the first line.  But yes, some may not.

jamiesnape, 2018-09-18:
Exactly.

EricCousineau-TRI, 2018-10-21:
Resolved since #9578.

Other todo items tracked by #9599 and related issues.

closed 2018-10-21 by EricCousineau-TRI