RobotLocomotion/drake #7915: doxygen for pydrake?

RussTedrake, 2018-01-31:
My current workflow for writing code in pydrake is a bit cumbersome... I normally read the API in the c++ doxygen, then have the pybind code open to see what pieces of the API are currently wrapped (and how).  What do we want this to look like in the future?  Should we generate a separate doxygen page for pydrake?  Should we be trying to make the docstring workflow the preferred approach?

\cc @EricCousineau-TRI

jwnimmer-tri, 2018-01-31:
See also #7914.

EricCousineau-TRI, 2018-01-31:
For writing the code, can I ask what environment you are using? Are you writing in plaintext? Or writing it in IPython?

One thing I haven't investigated is what `pybind11`s overloads look like in Python docstrings - my guess is that they've thought it through, but I have not yet investigated it.

Would you like me to prioritize this? (It would take a few days of focused time, either for manual porting (which I do not prefer, but would be a starting point) or the route of automation (to stabilize / minimize duplication).)

That being said, would it be OK if we consolidate discussion to #7914 or this issue?

RussTedrake, 2018-01-31:
Closing in favor of #7914.

closed 2018-01-31 by RussTedrake