RobotLocomotion/drake #7916: pydrake: Make a function to dump debug (support) information

EricCousineau-TRI, 2018-01-31:
Follow-up from #7903:

It'd be useful to have a one-stop function, first in pydrake, later in C++, that can dump things like Drake version, platform, etc.

As a first step, this method should be able to query `VERSION.txt` (if it exists) using `FindResource` to trace version information.

jamiesnape, 2018-01-31:
`VERSION.TXT` is not actually below the resource root (and I do not think it should be), so you may prefer to skip the first step. `VERSION.TXT` is really just an interim fix since the class is starting so soon.

EricCousineau-TRI, 2018-01-31:
What would a better step be? Have Drake generate something in source on `install`?
This seems like it'd be easy to have a "devel" version for Python, but not so for C++ - but I'd like to focus on Python first.

jwnimmer-tri, 2018-01-31:
The release engineering for compiled-in version badging can be tricky.  For now, a small human-only use python method that opened `/opt/drake/share/doc/drake/VERSION.TXT` and returned its contents (or some "unknown" text) would probably go a long way?

jamiesnape, 2018-01-31:
I agree. We do not need to overthink this for now.

EricCousineau-TRI, 2018-12-08:
Effectively resolved via #7903.

closed 2018-12-08 by EricCousineau-TRI