RobotLocomotion/drake #7920: Size of RobotLocomotion.github.io repo has exploded again

jamiesnape, 2018-01-31:
1.67 GB currently, documentation jobs may start timing out at some point soon.

FYI @stonier.

jamiesnape, 2018-01-31:
To me, looks like we could still tweak some more Doxygen settings. Small code changes are still inducing large numbers of changed generated documentation files.

RussTedrake, 2018-02-01:
is it the size of the doxygen content or the compilation time of doxygen that is the issue?

jamiesnape, 2018-02-01:
Cumulative size of the lots of surprisingly large diffs. Makes the clone of the repo slow. Generation time is fine. Strictly a byproduct of using a git repo.

sherm1, 2018-02-01:
Is it possible to nuke that repo altogether when it gets big and start fresh? (Assuming it just contains machine-generated content.)

jamiesnape, 2018-02-01:
Yes, we keep doing that every few months already. This is as much a reminder for me to do that soon, but equally it would be nice to fix the root cause.

jwnimmer-tri, 2018-02-03:
Possibly worth solving root cause in concert with #7791.

EricCousineau-TRI, 2018-12-03:
Given that this is not a huge pain point and not actively working on this, will close for now.

closed 2018-12-03 by EricCousineau-TRI