RobotLocomotion/drake #7944: Mutable arguments of type EigenPtr do not allow resizing

amcastro-tri, 2018-02-02:
Consider a method with signature:
```c++
CalcSomeMatrix(EigenPtr<MatriX<T>> output_matrix) {
  DRAKE_THROW_UNLESS(output_matrix != nullptr);
  if (output_matrix->rows() != 4 || output_matrix->cols() != 3) {
    output_matrix->resize(4, 3);  // This fails at runtime!!!
  }
  // Perform computation safely knowing output_matrix now has the right size...
}
```

The idea here is that we'd guarantee users of these methods that if they provide a matrix with the wrong size, then we'd resize it for them. However, per Eigen's docs, the `resize()` method is only valid for plain matrices/arrays, i.e. `Matrix::resize()` and `Array::resize()`. Since `EigenPtr::opertor->()` returns an `Eigen::Ref`, we cannot resize our input argument, even when it is meant to be mutable.

amcastro-tri, 2018-02-02:
cc'ing @soonho-tri, @EricCousineau-TRI since you guys worked with this class before.

jwnimmer-tri, 2018-02-02:
That `EigenPtr` forbids this is consistent with its intent.  An `EigenPtr` / `Eigen::Ref` means that you're going to scribble on the argument but not resize it.  The purpose of an `EigenPtr` is that the user could have passed in one of many concrete types of Matrix (such as a fixed-size one), but they all can convert into an `EigenPtr`.

I suspect your function should take `MatrixX<T>* out` instead of `EigenPtr`.  The `MatrixX<T>*` means you can call any method on it, including resize, but forces callers to pass a dynamically-sized Matrix... which they apparently already were, if you're planning on being able to resize it.

So "working as designed".  I'll leave the issue open in case you want to use it to track a documentation repair to avoid this confusion in the future.  Otherwise you can close it.

PR #7957 (2018-02-06): Adds clarification on the usage of EigenPtr
Updates `EigenPtr` doc per discussion in #7944.

Closes #7944.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/7957)
closed 2018-02-06 by amcastro-tri